J2Kbr/TitanTwo #5: My T2 keeps rebooting

pablogroup, 2016-09-28:
This is 100% reproducible when I trigger two different blocks of code in my script. It's weird because it compiles perfectly and this code is unchanged since T1 and has always worked for me. 

The first block begins on line 768 ("Walk option #3"). Basically if you double tap the shoot button (LT or L2), the Titan Two crashes and restarts. 

The second block begins on line 1012 ("Sniper Mode"). To replicate this, hold the ADS button (LT or L2), then press the X button on PS4 (or the A button on XB1). 

I use similar (if not identical) code patterns all over the place so I don't know why only these two are causing the unit to crash. 

Anyway here's the code:

`#pragma METAINFO("Rainbow Six Siege - XIM4 & Titan Two Mega Script", 1, 0, "pablogroup")

#include <ps4.gph>

//@@@@@@@@@@@@@@@@ RAINBOW SIX SIEGE: XIM4 + TITAN TWO MEGA SCRIPT @@@@@@@@@@@@@@@@ 

/* 

// IMPORTANT
//
// ----------------------------------------------------------------------------------------------------------------

This script is designed exclusively for: 
- Titan Two (as it is too big to fit within the limited space of a Titan One), combined with;
- XIM4

It is optimized for the following hardware configurations: 
- Playstation Navigation controller + Logitech G502 mouse (due to the limited 
  button count on the Nav, the G502 is essential if you are using a Nav). 
- Keyboard + standard mouse (any mouse will do, however my recommended setup 
  assumes that your mouse will have forwards and backwards buttons on the side, which
  99% of gaming mice do). 

KEYBOARD USERS VERY IMPORTANT NOTE: If you are a keyboard user, ensure that you change the variable keyboard_switch to "TRUE". 

// PREAMBLE
//
// ----------------------------------------------------------------------------------------------------------------

This script was originally designed for the Playstation Navigation controller and a Logitech G502 mouse. One of the biggest challenges of using this setup is that it only contains about 8-10 buttons that you can comfortably reach in the heat of the action - "comfortably" being the key word. To work around this limited button count, you will notice that I have regularly doubled up on buttons, used modifier keys, and in general I have made very creative use of the little that I have to work with - for example, double tapping sprint will activate weapon swap. So when you see the screenshots of my button mappings, they may look a bit unusual at first but it will all make sense once you've tried them for a while. I must emphasize that a lot of this setup has been designed to work well across the many menus and features. This is a complex game with a vast number of control scenarios, so don't just run around for a brief moment to test them - be sure to try things like vaulting, rappelling, navigating the various menus, and also test this for a number of different gadgets before you decide whether this works for you or not. If there's an unusual button mapping, don't be too quick to conclude "it still makes no sense", because it's probably designed for a menu or a scenario that you haven't even thought of. So when you first try it, I encourage you not to change any of the recommended button mappings, no matter how unusual they look at first. 

Being forced to work within such limits forced me to be creative and resourceful. This meant that when it came time to doing the keyboard port of this script, it was not only easy, but I found that I ended up making improvements that I otherwise wouldn't have thought of if I had started designing this script for keyboard users from the beginning. 

This is the product of over 600 hours of play testing, and 9 months of scripting and fine tuning. I am extremely happy with the results and I hope you will be too. 

// KEY FEATURES - SUMMARY
//
// ----------------------------------------------------------------------------------------------------------------

- Easy Sprint: simply push forward on the analogue stick to sprint. Seems easy enough, right? The problem with this is, because this game has so many complex and interrelated control systems, it causes all sorts of inadvertent issues. For example, sprinting will cancel the reload animation - it's annoying having to stand perfectly still in order to do anything! Also, you won't be able to swap weapons while sprinting, and it will cancel the placement of gadgets. The list of all the issues it caused was huge and I encountered problems pretty much everywhere I looked. The good news is I have managed to find solutions to literally every one of these problems! You would not believe the amount of testing that I had to put into this...

- Toggle Lean: this allows you to let go of the ADS button while aiming. This free's up your hand, which means you can strafe more aggressively to get the peeker's advantage. On the PC version of this game you are able to toggle-lean, so if you've ever been envious of how aggressively PC gamers can play because they have so many more options, you now have this advantage!
  
- A more robust leaning system: leaning just wasn't as smooth as I wanted it to be. There is a bug in the game that prevents you from leaning if you click Melee or Sprint too quickly. I fixed this by forcing a wait time on both the Melee and Sprint buttons, regardless of how briefly they were pressed. Overall, leaning feels much, much more robust and reliable now. 
  
- Primary Gadget Toggle Hold: IQ and Pulse's scanners, as well as Montagne's shield are very difficult to use because you need to hold down the primary gadget button the entire time they are deployed. Now you can tap a button and this script will automatically hold it down for you!
  
- Automatic Actions: This allows you to automatically place barricades, reinforce walls, and basically anything that requires you to hold down the Reload button. This also allows you to automatically initiate a rappel (so you don't have to hold down the Vault button for as long to initiate it). 

// OTHER FEATURES - IN DETAIL
//
// ----------------------------------------------------------------------------------------------------------------
- Easy Sprint: simply push forward on the analogue stick to sprint...no more buttons!
- Reload, swap weapons and use gadgets while "sprinting". Most Easy Sprint scripts don't support this, 
  so you need to be standing perfectly still to do anything.
- Automatic actions: are you sick of having to hold the Reload button to place barricades and to perform 
  other actions? For Nav users, tap the Vault button while holding Reload and this will happen automatically 
  (do the reverse to automatically initiate rappelling - hold Vault and tap Reload). For keyboard users, press 
  the "Interact" button (F key). 
- Includes three walk modes: 
[... 1312 lines cut ...]
         v = (v * s) / 100.0;
      }
      if(v > 100.0) v = 100.0;
      else if(v < -100.0) v = -100.0;
      set_val(i, v);
   }
   return;
}

fix32 midpoint(fix32 v, fix32 m) {
   if(v <= m) return((v * 50.0) / m);
   return(((50.0 * (v - m)) / (100.0 - m)) + 50.0);
}

uint8 getLastButtonPressed() {
	uint8 i, last = BUTTON_1;
	
	for(i=BUTTON_2; i<=BUTTON_21; ++i) {
		if(time_release(i) < time_release(last)) {
			last = i;
		}
	}
	return(last);
}
`

pablogroup, 2016-09-29:
It occurred to me this might have something to do with the use of the combo_restart function. 

I've used this in three places in my script - two of them are the two blocks of code that are causing the T2 to crash.

pablogroup, 2016-09-29:
Ok I'm pretty confident it is the combo_restart function causing this. I changed both instances to combo_run and it's fine now. 

Interestingly the third instance of combo_restart on line 833 doesn't cause a crash (look for the section titled "AUTO-ACTIONS").

pablogroup, 2016-09-29:
I have another thought - the one that worked has a wait time of 7000. The ones that crashed have very low wait times. I suspect it's the rapid cycling through the short wait times that's causing it. Anyway just an idea, one possible avenue for investigation...!

hhezart, 2016-10-07:
UPDATE: I used to get very frequent PS4 device disconnects and occasional reboots.   However for the past 3 days (12+ hours of gaming), I've disconnected the PROG (Mac/Fusion/Win10) during gameplay and not once I've experience an issue.  I've added the SD back and still everything is fine.

J2Kbr, 2016-10-09:
Thanks for the update. I will close this bug tracker then. Please post an new comment if the problem returns.

closed 2016-10-09 by J2Kbr