J2Kbr/TitanTwo #15: T2 not switching off when console in standby mode

pablogroup, 2016-10-06:
Not sure if this can be fixed but I thought I would log this anyway. 

Setup: T2 + XIM4
Console: PS4, with standby mode setup (so it doesn't go into a complete power off state)

Here's what happens: 

1) Put PS4 into standby mode, T2 turns off completely - good, as expected

2) A few moments later, T2 turns back on again, and is flashing red - not good

Is it possible for T2 to sense that the console is in standby and turn off completely?

prototypeGtuner, 2016-10-07:
There is an option to turn off USB ports when the PS4 is in stand by mode, did you check it?

pablogroup, 2016-10-08:
Oh I didn't realize I could do that. However what if you wanted to leave the USB ports on? This is still pretty jarring to look at: https://i.imgur.com/sFjHH9u.jpg

J2Kbr, 2016-11-06:
E4 is out-of-memory. Possible because XIM4 was in full pass-thru mode with PS4 and the PS4 stopped receive data packets, causing the Titan Two to buffer all data coming from XIM4 until out-of-memory.

The issue here is, Titan Two thinks there is a live PS4 connection. I will check if there is any mean to detect this specific condition and force quit from full pass-thru mode.

J2Kbr, 2016-11-06:
Quick update. Using the new PS4 controller everything works as expected while in rest mode and powered USB. No E4. Going try with an Hori controller, as this is the controller XIM4 emulates when connected on PS4.

J2Kbr, 2016-11-06:
Alright, another update.,  With Hori controller everything is working as well. Perhaps the changes made on the PS4/DS4 code (for the new PS4 controller) fixed the reported issue. Unfortunately I can't test with XIM4 at the moment. After the FW 0.77B release, please test with XIM4 and if didn't work, comment below to reopen.

closed 2016-11-06 by J2Kbr