J2Kbr/TitanTwo #17: E4 Error / Chat Headset / XBox One Controllers

iamgithub99, 2016-10-11:
Whenever I plug my standard Xbox One chat headset into the Elite or standard controllers directly or through the Chatpad, the T2 throws up an E4 error with all lights lighting up red.

See following link for additional info:
http://www.consoletuner.com/forum/viewtopic.php?f=23&t=4811

J2Kbr, 2016-11-06:
Another reference related with this issue:
Xbox One audio - static sound
Ref: http://www.consoletuner.com/forum/viewtopic.php?p=36273#p36273

J2Kbr, 2016-11-06:
E4 due headset connection is fixed on firmware 0.77 Beta. It was also made other improvements on the code that handles the audio data.

Tested playing TitanFall for about 1 hour, hearing other players and speaking a little bit.  Also performed tests like unplug/plug the headset in the middle of the match, after 5 to 10 seconds the audio and mic were working again. Controller used: Elite, but the results should be the same on all XB1 controllers.

closed 2016-11-06 by J2Kbr