J2Kbr/TitanTwo #18: Support for CTHULHU chip

pablogroup, 2016-10-12:
Back in the day, a lot of people with the Madcatz Fighting Stick installed the Cthulhu chip to make their sticks cross console compatible. 

http://forums.shoryuken.com/discussion/100475/rj-45-multi-console-cthulhu-arcade-stick-tutorial-ver-2

It seems like this is now causing a problem with T2. 

I have a PS3 fightstick with the Cthulhu chip, and it doesn't work on my XB1. It would be really good to be able to get this working otherwise it's a $300 piece of plastic:(

J2Kbr, 2016-11-06:
Please provide more information on this chip.

When you connect your fight stick  on Titan Two, does the LED port turns green? or remains red?
The CTHULHU chip has PS3 and XB360 modes? if yes, what happens if you use in XB360 mode?

pablogroup, 2017-02-01:
The update released on 21 Jan works perfectly with the Cthulhu chip! Closing

closed 2017-02-01 by pablogroup