J2Kbr/TitanTwo #19: Titan Two stop responding randomly on 0.75

amezibra, 2016-10-12:
since the upgrade to 0.75 and running a script that is fully working on previous version ,
I have random freeze where the controller become unresponsive until unplug/replugged.
T2 keep sending last input to console until controller is unplugged.

the freeze can happen twice in 10 minute and not happen again for 2 or 3 hours..
the used script is the destiny one i put on the T2 GPC repo.

hhezart, 2016-10-12:
It happened to me 3 times within an hour last night. In one instance my script was disabled and wasn't running.

amezibra, 2016-10-20:
update:

0.73: issue not observed
0.75: issue observed
0.76: issue not observed

so whatever has been done in 0.75, has been fixed in 0.76.

J2Kbr, 2016-11-06:
Thanks for update this bug report, confirming 0.76 solved the issue.

closed 2016-11-06 by J2Kbr