J2Kbr/TitanTwo #21: USB cable connectivity

digitalzombie, 2016-10-29:
While testing the unit with different micro USB to USB cables I have found out that the unit worked only with the cable that it was shipped. 3 other micro USB to USB cables that I have laying around which also work with Titan One, does not seem to work with this device. Gtuner does not detect the unit with the other cables.

digitalzombie, 2016-10-29:
Other USB cables which provide power but no connectivity actually allow Titan Two to connect to Gtuner IV for a split second and then the device gets immediately
[image]
 disconnected.

omasseau, 2016-10-29:
I also had connectivity problem but with the USB cable coming with the Titan Two.

When using it with my PS4 (Xbox 360 controller connected on Input A and PS4 controller connected on Input B), I was disconnected every 30s with the output led becoming red.

I replaced it with the USB cable of my Nexus 7 and everything works fine now.

J2Kbr, 2016-11-06:
Correct. Most connectivity issues reported so far are related with low quality USB cables. The one provided with Titan Two will be replaced in the next production batch. A good quality cable is important specially for the PROG connection, as this is a high speed USB link (480Mbps) and any external interference can lead to disconnections.

As this issue is not related to the firmware, I am closing.

closed 2016-11-06 by J2Kbr