J2Kbr/TitanTwo #22: Dual Shock 3 Not sending data over wired connection.

MaxRat08, 2016-10-31:
When connecting a DualShock 3 to the Titan 2 device it does not transmit any controller data. In the device monitor tab you can see that the controller is connected, but no button values are reported. All four lights on the controller blink together like the controller is charging but turned off.

J2Kbr, 2016-11-06:
Firmware 0.77B received couple changes in the Dualshock 3 controller driver (mostly to make it compatible with FragFX). As I couldn't reproduce this issue with the DS3s I have around, please ask to reopen this bug tracker if the changes did not fixed the described problem. Thanks.

MaxRat08, 2016-11-07:
I have tested both of my controllers with the new firmware and neither turns on, they just blink like they are charging (not the fast connecting to bluetooth blinking). The model numbers are CECHZC2U with a small A2 for the black one, and CECHZC2U with a small A1 for the blue one. Both work perfectly with my console.

J2Kbr, 2016-11-07:
Reopen.

J2Kbr, 2016-11-15:
Cause of this bug was found and fixed on Titan Two firmware 0.78B.
The issue was related with Bluetooth pairing procedure without an BT Module installed.

closed 2016-11-15 by J2Kbr