J2Kbr/TitanTwo #23: Xbox 360 wireless controller rumble issue

prototypeGtuner, 2016-11-01:
Playing on Xbox 360 and PS4, most of the time there is no vibration for the Xbox 360 wireless controller.

Xbox 360 < Titan Two < INPUT A > Xbox 360 wireless controller    //
                                        INPUT B > Xbox 360 wired controller      // or vice versa

PS4 < Titan Two < INPUT A > Xbox 360 wireless controller     //
                               INPUT B > DS4 controller                         //or vice versa

J2Kbr, 2016-11-06:
I did few tests and for me it seems to be working as expected (Official Xbox 360 wireless controller). Perhaps there is an small difference in rumble intensity when playing on PS4 with an XB360 controller. To check that please play any Xbox 360 game to fell if the rumble is more noticeable. Thanks.

prototypeGtuner, 2016-11-07:
Playing on Xbox 360 with Xbox 360 wired controller in INPUT A and DS4 controller in INPUT B, the rumble is more noticeable on the DS4 than the Xbox 360 wired controller. Swapped DS4 to Xbox 360 wireless controller in INPUT B, the rumble on the Xbox 360 wired controller are noticeable but there is no rumble on the Xbox 360 wireless controller . With Titan Two up to date powered from the output port or the prog port the result is the same.

prototypeGtuner, 2016-11-22:
Regarding this test:
http://www.consoletuner.com/forum/viewtopic.php?f=27&t=5222&p=39413#p39413

For me the DS4 have stronger rumbles than the Xbox 360 wireless controller, maybe it's why sometimes there isn't vibration on the Xbox 360 wireless controller.

Closed issue.

J2Kbr, 2016-11-22:
Thanks for the update. I will be replying on the topic linked shortly.

closed 2016-11-22 by J2Kbr