J2Kbr/TitanTwo #30: (Feature Request) Add Steam Controller Support

MaxRat08, 2016-12-01:
The title says it all. Would be the ultimate controller for people who want one controller for everything they use.

pablogroup, 2016-12-02:
Apparently Valve have been terrible about developer support. OBSIV the owner of XIM wanted to support it too but poor documentation and communication forced him to put this on the back burner. Shame really. That company makes too much money on steam to care I guess.

J2Kbr, 2017-10-31:
Note: In the past days I've been working on the support for the Steam controller. For my surprise it can already be used, the Titan Two recognizes it as keyboard and mouse, which could be perfectly used to play games after with some buttons remapping and sensitivity adjustments. The Steam controller also have a proprietary interface which works more like a gamepad, I am currently reversing engineering of this proprietary protocol to extend the support to this interface as well.

MaxRat08, 2017-10-31:
Awesome news! Just out of curiosity, (assuming reverse engineering the gamepad interface goes smoothly) how were you planning on mapping the two touchpads on the controller? Are they going to be a dpad and joystick, or were you going to have them map into the same registers as the DS4's touchpad?

J2Kbr, 2017-11-02:
As the Titan Two have two pairs for point inputs (`POINT_1X`, `POINT_1Y`, `POINT_2X`, `POINT_2Y`) I believe we will end using those. Of course, any mapping can be done via script or input translator, accordingly the user needs.

MaxRat08, 2017-11-02:
Cool, that seems like the better way in my opinion. Definitely looking forward to this!

J2Kbr, 2018-01-25:
Full support (wired, wireless) for Steam Controller added to firmware 0.98B-5.

closed 2018-01-25 by J2Kbr