J2Kbr/TitanTwo #31: (Feature Request) LCD Smartie support

prototypeGtuner, 2016-12-02:
http://www.consoletuner.com/forum/viewtopic.php?f=22&t=4094

J2Kbr, 2017-10-31:
I gave some thought to this feature. This is a really nice idea, but I don't see be useful for users in general, as it would require the purchase this specific the LCD display. And, for the same purpose, we have planned the integration with smartphones for device configuration and status feedback, which will offer a much nicer user interface than just the BW LCD display. In any case, users will still be able to create a interface with this or others displays with the Titan Two open API.

closed 2017-10-31 by J2Kbr

prototypeGtuner, 2017-11-12:
Ok i understand that it will not be useful for users in general because it require a specific hardware. The API opens the doors for a bunch of applications and i keep hoping because LCD Smartie is open source and the difference with the smartphone is the real-time display! It's like having the Output Panel on an external display!! That would be possible! Do you have a time line for the API?