J2Kbr/TitanTwo #40: Implement outstanding XIM scenarios

PablosScripts, 2017-03-08:
Placeholder.

Missing a few, e.g. authenticating reliably via T2 instead of XIM using a non-ps3 pad.

Boosting reliability on PS4.

etc.

J2Kbr, 2017-03-11:
Hi Pablo. I am doing some tests with XIM4 (latest firmware) - right now checking on PS4. I am making an list of troubleshooting procedures and suggestions - so far there are two main issues I could detect:

1. Cables
2. Lack of Power

Please give me some problematic usage/connections, so I can reproduce and fix. Thanks.

Despite that, some users are reporting purple light with PS4 - I am unable to reproduce that. Not sure if the recent changes on the firmware (not released yet) solved this issue, or if it is something else on user's end (i.e. cable/lack of power). When you use XIM4+TitanTwo on PS4, do you get purple lights?

J2Kbr, 2017-03-11:
Okay, purple lights are solved with the latest XIM4 firmware. Please disregard the above question on this matter. Thanks.

GarethWright, 2017-03-13:
The unreleased one or  20170103?

J2Kbr, 2017-03-13:
I testing with the version from 20170103.

GarethWright, 2017-03-13:
Roger that. I've bought a powered hub to sit between the XIM and the keyboard and mouse as I'm still getting purple lights on XB1. I'll test that tonight.

closed 2017-08-10 by J2Kbr