J2Kbr/TitanTwo #41: Start of GTuner with connected T2 with SD-card & my script on it results in connection failure

J2Kbr, 2017-03-08:
Original Issue: https://github.com/J2Kbr/GtunerIV/issues/83

**Scachi commented on Nov 16, 2016**
> Something strange is going on here again with the Titan2 with SD-cards.
> Not sure at all what is causing this so I just write down every information I can think of that may help to track this problem down.
> 
> I think this problem started for me during creating a script for Magicka2 for the PS4 and drag&drop it onto slot5 for the first time (used the test & debug button until now).
> When I dropped it onto slot5 the T2 disconnected in GTuner and the device&slot panels where greyed out.
> I did pull one cable of the T2 and replugged it in trying to get it connected again, without any success.
> Resulting in the log msg: 
> ```
> Connect Fail: Error code: 2, Device: TITAN TWO [ED36FFFF713285B83953343638362020FF0510B9]
> ```
> 
> I ended up with just the T2, without SD-card, and the USB prog cable and the T2 connected successfully again in GTuner.
> This CFG#1 (T2+prog cable NO sd-card) seems always to work. No matter if I  start GTuner up first and connect the T2 with the prog cable after that or the other way around.
> 
> Okay, using CFG#1 and having GTuner running with the T2 connected I have a high chance (does not always work on the first try) that I can use the SD-card now by putting it in. Showing in the output panel:
> `Micro-SD Card: Inserted (1.85GB)`
> and the content show up correctly in the Memory Slots (Micro SD) panel.
> If I close GTuner (leaving the SD-card plugged in and only the prog port connected) and restart it, the Titan2 disconnects again. 
> Outputpanel:
> ```
> System: Windows 10 (winnt 10.0.14393, x86_64, i386-little_endian-ilp32, LibGL)
> OpenGL: C:/WINDOWS/SYSTEM32/opengl32.dll
> Locale: German, Germany
> Gtuner: IV 0.94.1.1320 DE:Set - Error loading language
> Current Dir: C:/Users/scach_000/Documents/GtunerIVBeta/GtunerIVBeta
> Data Dir: C:/Users/scach_000/Documents/GtunerIVBeta/GtunerIVBeta
> Working Dir: C:/Users/scach_000/Documents/GtunerIVBeta
> Device Selector: added 1 device(s), total of 1 device(s) detected
> Connect Fail: Error code: 2, Device: TITAN TWO [ED36FFFF713285B83953343638362020FF0510B9]
> ```
> If I leave GTuner up and running and the SD-card in the Titan2 and unplug+replug the prog cable it has a chance that the T2 stays connected and shows up all slots content correctly.
> If it fails it disconnects the T2. Unplug/replug the prog cable a few time works to get it stay connected and showing up the slots content correctly.
> 
> When I do all this with a power supply connected to the T2, unplug&replug the T2 has now chance to get it connected to a running GTuner . Removing the power supply or let the T2 enter the P mode
> a quit it again seems to reset the memory/device and there is the chance again to get it connected to a running GTuner.
> 
> Removing my Magicka2 script from Slot 5 seems to completely  solve this problem and I can start up GTuner with the T2 and SD-card already connected to my pc.
> 
> Drag&drop the Magicka2 script to slot 1, 2 or 3 has a chance get the T2 disconnected as long as a sd-card is inserted. (The sd-card slots may has to be filled with scripts).
> But drag&drop it to a sd-card slot seems to have a higher chance to get the T2 disconnected.
> 
> Wild guesses:
> I wonder if drag&dropping any script to a sd-card slot always start the init function of the script no matter if this script is in a active slot or not. Or when GTuner starts and reads the slots content is somehow starts the init functions or is doing some pmem stuff it shouldn't do at this stage.
> As using the sd-card with Magicka2 on it in the T2 without using GTuner seems not to cause any problems.
> 
> To be fair, my Magicka2 code is a complete&complex mess as I am using a lot of pointers and funny functions with for loops over arrays and it is not completed and tested yet with all it functions.
> I may have tons of bugs in there.
> 
> The Magicka2 script is attached.
> [Magicka2_Scachi.zip](https://github.com/J2Kbr/GtunerIV/files/594617/Magicka2_Scachi.zip)
> 
> Sorry for this long post.

**Scachi commented on Nov 16, 2016**
> Hm, putting the script into a internal slot and starting GTuner with the T2 prog cable already connected works.
> Putting the sd-card into the T2 (magicka script still only in slot 2, not on the sd-card slots) and starting Gtuner gets the T2 disconnected in GTuner (greyed out Memory panels).

**Scachi commented on Nov 17, 2016**
> This still happens with the new firmware and gtuner software.
> 
> Really, I have no idea how to reproduce it.
> I tried it with another SD-card. Formatted it from within gtuner and put all the scripts to the slots I have with the other card. And no problem drag&dropping anything and no other problem.
> It looks like it has nothing to do with the Magicka2 script.
> I can't believe I am the only one with this problem. Starting with the T2 I was the only one with that E1 problem in its early days.. maybe it still has something to do with my specific T2 device ?

**Scachi commented on Nov 24, 2016**
> Using the T2 without an external power supply didn't help.
> Today the problem is back. Starting gtuner with prog cable connected and sd-card inserted, the T2 gets always disconnected.
> Without the SD-card I can start gtuner with the prog cable connected. SOmetimes I can get the SD-card slots display correctly when I insert the sd-card after gtuner is already started and the T2 connected before I insert the sd-card.
> 
> So, no, it doesn't look like a heat problem because of the external power supply or related to the external power supply.

**Scachi commented on Jan 28**
> New reports from other T2 owners with the same issue:
> http://www.consoletuner.com/forum/viewtopic.php?f=27&t=5665

J2Kbr, 2017-03-09:
Hi Sachi, I am currently investigating issues related with sd-card interface and already did some optimizations on the API, such as disable the support for SDIO (ie. use the sd slot for bluetooth, GPS, etc) which is not needed for the Titan Two. Not sure yet if theses changes will help on the issues described.

To start, just a comment on this  consideration:

> I wonder if drag&dropping any script to a sd-card slot always start the init function of the script no matter if this script is in a active slot or not. Or when GTuner starts and reads the slots content is somehow starts the init functions or is doing some pmem stuff it shouldn't do at this stage.
As using the sd-card with Magicka2 on it in the T2 without using GTuner seems not to cause any problems.

No, the script is not executed when drag-and-drop it to an memory slot, or when Gtuner IV is reading the memory slot contents. Definitively the issue is not related with the script bytecode data.

I tested again the scenario with the Magicka2 script and -unfortunately- couldn't reproduce the issue.  I have, however,  noticed an performance degradation over-time of the sd-card operations, which causes issues like errors when dragging-and-drop scripts to sd-card memory slots, very slow loading of an memory slot located on sd-card, connection error 2 with Gtuner IV. To fix that I simply re-format the sd-card using Gtuner IV - if an error occur when formatting, I try an second time with success 100% of the time. After re-formatting everything works as expected and fast for an while, when the problems starts again.

I have not yet identified what is causing this issue. My actual guess is the FS table, or some other control data structure, is getting in an state not optimal, increasing the time needed for the operations on the sd-card. This morning I've looked at the sd-card API for any function related with close/finalize the control data structures used to access the sd-card, but couldn't find any. I've also checked the examples of use the API, and it seems I'm doing everything the right way on Titan Two.

For reference, these are the API in use for sd-card access:
http://asf.atmel.com/docs/3.21.0/uc3a3_a4/html/group__avr32__drivers__mci.html
http://asf.atmel.com/docs/3.21.0/uc3a3_a4/html/group__group__avr32__services__fs__fat.html
http://asf.atmel.com/docs/3.21.0/uc3a3_a4/html/navigation_8h.html

--
I recently re-formatted my sd-card and my plan now is use some diagnosis tool on it after the issues come back, Hopefully this will tell me what is wrong with the sd-card.

prototypeGtuner, 2017-03-10:
I don't know if it's a SD card related issue but each time i plug or unplug the PROG cable into the PC (with or without the power supply) there is a disconnection on the console side. And recently i had a problem with a script in the slot 8 that was not over writable. I had to format to be able to use the slot 8.

J2Kbr, 2017-03-12:
@prototypeGtuner, this is not related with sd-card. However, I am aware of this issue and the firmware 0.84B will fix this.

J2Kbr, 2017-05-14:
Since the latest firmware update I've not noticed any problem with SD-cards. If is not the case for you, please reopen this issue.

closed 2017-05-14 by J2Kbr