J2Kbr/TitanTwo #42: Titan Two's lightbar intensity

prototypeGtuner, 2017-03-10:
When i change the intensity from the device configuration the Titan Two's lightbar intensity reset to something like 50% after each step in the scrolling bar. The intensity change but reset after some ms to 50%. Same behavior on the DS4 LED.

prototypeGtuner, 2017-03-10:
Lightbar option is set to "player indicator or Script"

prototypeGtuner, 2017-03-11:
Works fine when connected thru prog port only but not when connected to the PS4. For both T2 and DS4 leds.

J2Kbr, 2017-03-12:
Actually this is the expected behavior. Please note the PS4 is the only console that reports LED intensity and, for that reason, when using the Titan Two on the PS4 with "Player LED" enabled, the lightbar intensity should be configured on the PS4 settings -> Brightness of DUALSHOCK 4 Light Bar.

closed 2017-03-12 by J2Kbr