J2Kbr/TitanTwo #43: INPUT-C works but not INPUT-D

prototypeGtuner, 2017-03-12:
My keyboard works and the output led lights up on INPUT-C but don't works or lights up on prog port (INPUT-D).

prototypeGtuner, 2017-03-12:
There is a quick red light up when i unplug the OTG cable + KB nano receiver.

J2Kbr, 2017-04-06:
This week I dedicated some time investigating why INPUT-D is not working.

The Titan Two has two USB controllers, one is internal of the micro-controller (PROG/INPUT-D) and other is external, a dedicated Hi-Speed USB controller, which takes care of INPUT-A, INPUT-B, INPUT-C and OUTPUT. 

There are some challenges to integrate the USB data flow between the different USB controllers, specially for the full-passthru mode. Implement an solution for that will take more time than I was expecting.

J2Kbr, 2017-04-27:
INPUT-D (PROG port with OTG adapter) working on firmware 0.88B for HID devices (keyboard, mouse, joystick, fight stick, gamepad, steering wheel, flight stick).

closed 2017-04-27 by J2Kbr