J2Kbr/TitanTwo #44: Xbox One S Controller - Inconsistent FFB

J2Kbr, 2017-03-14:
Reference:
http://www.consoletuner.com/forum/viewtopic.php?p=43989#p43989

> Any event that causes my controller to vibrate will cause my controller to vibrate as expected but the controller will continuously keep vibrating a lot longer than its suppose to.

PablosScripts, 2017-04-01:
I may have some input into this. 

I have found something similar happening before when I've had too many devices connected to my T2. The vibration just never stops. Once I start unplugging more devices and there is a lower power draw, the problem goes away. I assume the AC power supply will fix this.

I find it's particular problematic with the PS4 controller as I think it has a higher power draw than the XB ones.

J2Kbr, 2017-04-06:
Issues with Xbox One S controller fixed on firmware 0.87B

closed 2017-04-06 by J2Kbr