J2Kbr/TitanTwo #45: Battery reports when using the Nacon Pro Controller?

prototypeGtuner, 2017-03-24:
Since i use the Nacon (awesome controller) i have a lot of battery reports saying the controller is empty while playing, or 25% or 50%. This controller should not have battery reports? When using the Titan Two with only the Nacon i get these battery reports and it is the same when a second controller is plugged, a fully charged DS4. I don't know if it's really a bug or the usual DS4 reports.

prototypeGtuner, 2017-03-24:
The second DS4 is always plugged in the Titan Two, for the speaker! But sometimes the speaker settings are not available.

It's awesome to have the Titan Two and the DS4 speaker on the table with the Nacon 2 or 3 meters away, it's like having a speaker in the Titan Two!

prototypeGtuner, 2017-03-24:
Everything unplugged, i plug the Titan Two in the PS4. AU time, i plug the DS4 in Input B. The PS4 says the DS4 is fully charged. Then i plug the Nacon in the Input A and the PS4 says the DS4 is empty! Looks like there's a small bug.

prototypeGtuner, 2017-03-29:
some more details.
I have issues when both DS4 and Nacon are connected, it appears as disconnections due to empty controller. I don't know which setup works better, plug the DS4 first, or the Nacon...

J2Kbr, 2017-04-06:
Battery charge level report of third-party controllers fixed on firmware 0.87B. Thanks for reporting.

closed 2017-04-06 by J2Kbr