J2Kbr/TitanTwo #47: Clear Memory Slot issue

prototypeGtuner, 2017-04-04:
I have a problem to clear an Input Translator from a slot.
The slot works ingame and i can clear the script but not the input translator.
I've done a Firmware recovery with no luck, how to format the Titan Two's internal memory slots?
https://www.consoletuner.com/forum/viewtopic.php?f=25&t=6037

prototypeGtuner, 2017-04-04:
After one other Firmware recovery i the slot have been deleted.

prototypeGtuner, 2017-04-13:
I don't know where is the problem but after drag and drop an other script in the slot 1 the Input Translator appears, after had been deleted! A video is better:
https://www.mediafire.com/?gg04zogxw3rhm8g
look until the end, during the time that nothing happens i'm dragging some script several times end at the end they appears sudenly when they should not!

prototypeGtuner, 2017-04-13:
After formate the SD card one more time everything works fine with normal scripts and when i put a script with "1 line art codes" it make the big mess again. And the problem with the Input Translator appears in the slot 1

#pragma METAINFO(" ̿ ̿̿'̿̿\̵͇̿̿\=(•̪●)=/̵͇̿̿/'̿̿ ̿ ̿  ", 1, 0, "function")

J2Kbr, 2017-05-13:
This issue seems to be Gtuner IV related, move to:
https://github.com/J2Kbr/GtunerIV/issues/143

closed 2017-05-13 by J2Kbr