J2Kbr/TitanTwo #49: Logitech G pro Gaming keyboard not recognized

amezibra, 2017-05-10:
T2 doesn't get any input from this simple script:

```
char current_key;
int key_ready;
 
init {
 
}
 
main {
    if (key_get() != current_key ) {
		key_ready = 1 ;
		current_key = key_get();
		printf("Key Code: 0x%02X", current_key);
 
	}
 
}
```

when i try to fetch the descriptor with T2 gtuner reload.

any idea?
here is a fetch from windows:

Information for device Pro Gaming Keyboard (VID=0x046D PID=0xC339):

Connection Information:
------------------------------
Connection status: Device connected
Device actual bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device is hub: No
Device address: 0x001B
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass   
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 Bytes)
0x046D	idVendor
0xC339	idProduct
0x1301	bcdDevice
0x01	iManufacturer   "Logitech"
0x02	iProduct   "Pro Gaming Keyboard"
0x03	iSerialNumber   "0123456789"
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x003B	wTotalLength   (59 Bytes)
0x02	bNumInterfaces
0x01	bConfigurationValue
0x04	iConfiguration   "U113.01_B0013"
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0xFA	bMaxPower   (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass   (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0041	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress   (IN Endpoint)
0x03	bmAttributes	(Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize   (8 Bytes) 
0x01	bInterval

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass   (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0077	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress   (IN Endpoint)
0x03	bmAttributes	(Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize   (64 Bytes) 
0x01	bInterval

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Logitech"
0x02   0x0409  "Pro Gaming Keyboard"
0x03   0x0409  "017136623135"
0x04   0x0409  "U113.01_B0013"

------------------------------

Connection path for device: 
Contrôleur d?hôte compatible xHCI USB
Root Hub
Pro Gaming Keyboard (VID=0x046D PID=0xC339) Port: 7

Running on: Windows 10 or greater

Brought to you by TDD v1.84.0, Dec 14 2015, 09:19:38

J2Kbr, 2017-05-12:
Was made some changes on the firmware (not yet released) to try fix this issue, not 100% sure as I don't have this keyboard to test. The new firmware will possible be out during this upcoming weekend, as an revision of 0.88B.

I am closing this issue, in case you find out it didn't resolved, please reopen. thanks.

closed 2017-05-12 by J2Kbr

amezibra, 2017-05-15:
tested last firmware now the keyboard is working perfectly 👍 
thanks