J2Kbr/TitanTwo #50: Titan Two Powered Jack issue with Xim4

amezibra, 2017-05-11:
scenario:
T2:
slot : no script
input A DS4 v2
input B xim4
output PS4 pro
prog PC
powered via jack 

Xim4:

output To T2
port 1 DS3 stock controller
port 2 usb hub dlink
port 3 unused

what happen is T2 goes into continuous boot loop until jack is removed.
this doesn't happen everytime but it does often especially if you unplug xim4 from input B and put it again.

console message is as follow  

Disconnect: Titan Two device has been disconnected
Device Selector: removed 1 device(s), total of 0 device(s) detected
Device Selector: added 1 device(s), total of 1 device(s) detected
Connect: TITAN TWO [ED36FFFF6B6E0E103953343638362020FF050ECE] connected
Disconnect: Titan Two device has been disconnected
Device Selector: removed 1 device(s), total of 0 device(s) detected
Device Selector: added 1 device(s), total of 1 device(s) detected
Connect: TITAN TWO [ED36FFFF6B6E0E103953343638362020FF050ECE] connected
Disconnect: Titan Two device has been disconnected
Device Selector: removed 1 device(s), total of 0 device(s) detected
Device Selector: added 1 device(s), total of 1 device(s) detected
Connect: TITAN TWO [ED36FFFF6B6E0E103953343638362020FF050ECE] connected

amezibra, 2017-05-12:
after further testing it is triggered but not always by plugging the xim4 usb cable to T2:B 
i have it stabilized and working but it definitely happens sometimes on cable swaps..

amezibra, 2017-05-16:
i think we can close this one as i identified it to a faulty cable.
there is some kind of usb cables who trigger this.

paname75, 2017-05-18:
not sure if it is related to your last update but i don't have this issue anymore with xim4

J2Kbr, 2017-08-10:
Perhaps this issue is related with #40.
Anyway, thank you for reporting and confirming this issue is solved.

closed 2017-08-10 by J2Kbr