J2Kbr/TitanTwo #51: Mic Audio Problem occurs after .87 Firmware for xim4 users on the Xbox one

Fragsnipa, 2017-05-14:
The latest firmware released april 7 .87B has giving xim4 users mic audio problems. You voice will sound very robotic and static. I have test titan 2 without the xim4 and the mic audio is fine but when they are both connected I get a very static voice.

https://1drv.ms/v/s!AjwPsHpvzSlKbPl59oSl5FT-f4Q

J2Kbr, 2017-08-10:
Quote from #63 (duplicated issue)
**Montly34:**
> Environment: XB1, T2 Firmware 0.92.1, XIM4 (latest Firmware), Logitech G13 & 502, Xbox controller
> 
> Steps to replicate:
> 
>     XIM 4 with Mice/Keyboard and Controller (Headset plugged in Headset Adapter & Controller)
>     XIM 4 connected to T2 either in Port A or B
>     Connect T2 into XB1
> 
> Expected outcome:
> Without any script loaded, i expect the same behaviour of XIM 4 as plugged directly to the Xbox
> 
> Actual outcome:
> XIM 4 Shows purple flashing lights
> No Input via Mice and Keyboard possible
> No Audo via Headset, Headset is not accepted by Headset Adapter (can´t mute it)
> Tested in both Inputs A & B

J2Kbr, 2017-08-10:
Here is a workaround to use XIM4 with Titan Two on Xbox One with controller headsets.

1. Xbox One controller with headset on INPUT-B;
2. XIM4 with keyboard/mouse and PS3 controller on INPUT-A.

This solves purple-light on Xbox One and allow use the headset at the same.

hiroshimus, 2017-09-21:
Hey J2Kbr,

The thing is that my current setup wouldn't allow me to connect the Xbox One controller on Input B. It would defeat the purpose of all the adapters and peripherals that I've bought and I really don't want to disconnect and reconnect anything at all.

Will there ever be a solution for this chat issue caused by Xim 4 and Titan Two?

J2Kbr, 2017-12-07:
After investigation on mic audio issues (not related with XIM4) the problem was fixed by requesting isochronous data earlier than what is specified by the bInterval parameter of the controller.

Related with XIM4, I have already observed the cause of the problem was the Titan Two inquiring XIM4 "faster" than what it is expecting. The solution described above -unfortunately- made the problem with XIM4 even worst.

The priority with the Titan Two is provide good audio for PS4 and XB1, primarily using the controller connected to the Titan Two. An attempt to fix the issue with XIM4 would revert the progress made.

There are two possible solution for XIM4 mic audio issue:

1. Find an away for the Titan Two "detect" the connected device is XIM4 and then apply a slower data request from the USB endpoint;
2. Changes on XIM4 firmware to allow faster polling interval for the USB isochronous endpoint.

On either case the solution depends on XIM4. As such, on the first opportunity, I will be communicating OBSIV about these finding.

closed 2017-12-07 by J2Kbr

hiroshimus, 2017-12-07:
Alright thanks, it's just that since T2 was supposed to support Xim 4 (including voice chat, unlike t1) I thought that voice chat had been tested using T2+XIM4 so this would've been communicated to all of us before buying a T2. I mean I would've probably still bought a T2, but you know what I mean.

One of the main reasons I bought  T2 is because it claimed that it supported voice chat using a Xim4. I think the T1 is great, but not being able to communicate with others while playing competitive games is very hard (I own a T1 as well).

Anyways, I created a post on Xim 4 forums where I mentioned what you said on this post, to see if they are gonna be able to help with this issue or not.