J2Kbr/TitanTwo #53: INPUT-A INPUT-B working inconsistency with some combination of input devices.

J2Kbr, 2017-06-07:
Reference:
https://www.consoletuner.com/forum/viewtopic.php?p=47167#p47167

oneswitch, 2017-06-14:
Tried resolving this with a regulated 7.5v 2amp power supply, but it did not improve the problem.

Swapping controllers did not resolve the issue.

I've been testing with (A = HORI SRWC PS2 controller through a daisy-chained PS2 to Xbox 360 adapter..... B = Sony PS4 controller).

oneswitch, 2017-06-15:
This is a picture of the set-up....

[image]

Perhaps it's something to do with the Xbox 360 adapter, but this is otherwise one of the best for compatibility with a number of combinations

[image]

Be happy to post you one if it helps. I'm still not 100% if it is that adapter causing the problems with the PS4 controller getting locked out. Strange one though.

Fragsnipa, 2017-06-20:
I can confirm this also, using the 360 controller with xim4 and using the xbone controller as the auth. I do experience of the seizing up on the xbone controller when I play destiny.  When I go in the menu to switch weapons. The xbone controller  or the keyboard and mouse that Im using with the xim4 will seize up.  And Im using a power usb hub so its seizing up not a power issue.

oneswitch, 2017-06-27:
With the latest Titan Two updates, I'm still getting this random problem of controllers locking up. This time trying with an official PS4 Dual Shock and an Ultimarc iPac (in joystick mode). 

Descriptors for the iPac are:
`[USB DESCRIPTOR]
12 01 00 02 00 00 00 08 09 D2 20 04 37 00 05 05 03 01 09 02 6D 00 04 01 02 A0 FA 09 04 00 00 01 03 00 00 04 09 21 11 01 00 01 22 4E 00 07 05 81 03 20 00 05 09 04 01 00 01 03 01 01 01 09 21 11 01 00 01 22 3F 00 07 05 82 03 20 00 05 09 04 02 00 01 03 01 02 01 09 21 11 01 00 01 22 34 00 07 05 83 03 08 00 05 09 04 03 00 01 03 00 00 01 09 21 11 01 00 01 22 88 00 07 05 84 03 40 00 0A 

[STRING 5 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 05 04 09 00 20 
20 03 55 00 6C 00 74 00 69 00 6D 00 61 00 72 00 63 00 20 00 49 00 50 00 41 00 43 00 20 00 32 00 

[STRING 5 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 05 04 09 00 20 
20 03 55 00 6C 00 74 00 69 00 6D 00 61 00 72 00 63 00 20 00 49 00 50 00 41 00 43 00 20 00 32 00 

[STRING 3 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 03 04 09 00 04 
04 03 37 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 00 00 4E 
05 01 09 05 A1 01 09 01 A1 00 09 30 09 31 16 00 F8 26 FF 07 36 00 F8 46 FF 07 65 00 75 10 95 02 81 02 09 39 16 00 00 26 07 00 36 00 00 46 07 00 75 08 95 01 65 00 81 02 05 09 19 01 29 20 15 00 25 01 35 00 45 01 95 20 75 01 81 02 C0 C0 

[INTERFACE 1 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 01 00 3F 
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 1E 75 08 15 00 25 65 05 07 19 00 29 65 81 00 C0 

[INTERFACE 2 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 02 00 34 
05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 03 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 

[INTERFACE 3 HID DESCRIPTOR]
00 00 00 03 00 00 02 01 81 06 22 00 00 03 00 88 
05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 01 95 03 75 08 81 01 C0 05 0C 09 01 A1 01 85 02 15 00 25 01 09 E9 09 EA 09 E2 09 CD 19 B5 29 B8 75 01 95 08 81 02 95 03 75 08 81 01 C0 05 01 09 00 A1 01 85 03 05 09 19 00 29 20 15 00 25 01 95 20 75 01 81 02 05 08 09 4B 91 20 75 08 95 05 C0 05 06 09 00 A1 01 85 04 05 08 09 4B 75 08 95 60 81 02 05 08 09 4B 75 08 95 60 91 20 C0 
`

I will try with some different USB cables, as it does seem quite fussy with them, more so than the T1. Any news on looking replicating this problem your side?

oneswitch, 2017-06-28:
Can't seem to pin it down. Sometimes it happens. Sometimes it doesn't. Doesn't seem to be any one thing that I am doing. Just times out on occasion and blocks the official PS4 controller.

J2Kbr, 2017-07-04:
I was able to reproduce this problem.
Issue fixed on Titan Two firmware 0.91B, to be released yet this week.

oneswitch, 2017-11-10:
This problem seems to have resurfaced, using an almost identical set-up as above. This time I'm using power via the PROG socket (from a RetroPie power supply, so should be up to the job).

Part way through playing, the other controller either stops responding at all (if you were only using one controller), or worse, controls might lock on if you're using both at the same time in "co-pilot" fashion.

J2Kbr, 2017-12-01:
The firmware 0.97B-2 reverted a power setting change made on the firmware you noticed this problem again, I expect this to be solved with the 0.97B-2, please comment if not. thanks.

closed 2017-12-01 by J2Kbr