J2Kbr/TitanTwo #54: Raspberry Pi / RetroPie: (in Xbox 360 output protocol) worked with T1 but won't with T2

oneswitch, 2017-06-14:
The T2 does not seem to be communicating well with a Raspberry Pi for some reason. The T1 worked perfectly, when set to Xbox 360 output protocol (much like when using it on a PC).

If I try the same with the T2, it works on a PC, but not well on a Raspberry Pi. It's as if the basic controls are being passed through, but all the script features are being ignored.

Also the adapter I'm using (PS2 to Xbox 360) constantly flashes all 4 LEDs as if it's failing on some kind of hand-shaking.

J2Kbr, 2017-08-10:
Related with the XB360 output protocol on Raspberry Pi. If I understood correctly, the controller works normally, just the scripting features don't. Is that correct?

In regards the PS2 adapter, several improvements were made since latest firmware updates. I expect this adapter to be working now, please confirm. thank you.

oneswitch, 2017-08-10:
:) 

The PS2 compatibility via a Brook is working well. Everybody's Golf worked perfectly with scripts taking effect. However the Brook has a problem (tried with Who Wants to be a Millionaire 2nd Edition) and that works with the shape buttons to skip the credits screens, but stopped working on the "press any button" screen). I'm asking them to look into this, as it's a compatibility issues likely to affect other games. Be great to get this fix working on the T1 too if possible?

The Raspberry Pi compatibility is still strange. I could only manage to get it part working by forcing into XB360 mode, and by connecting an Xbox 360 standard controller. The scripts would not work, and if I tried a PS4 controller attached on its own or along with it... that did not work at all. At best the Xbox 360 controller was passed through as a basic controller.

oneswitch, 2017-08-21:
Brook managed to fix the issue which you can see misbehaving here: [https://www.youtube.com/watch?v=Q2rE5HiSYEs](url)

Not sure what they did but I can ask. 

Wonder if it's related to the strange problems with UCR (and sometimes GTuner alone) and PS3 use.

oneswitch, 2017-08-21:
Problems remain with UCR through the Titan One into the Brook PS3/PS4 to PS2 adapter. The d-pad seems to be no good in Everybody's Golf (d-pad response is odd in a number of PS3 games too). Here, the d-pad seems to work fine in the menu, but when used for constant left/right/up/down movement, it's very, very slow on up/down or for left/right non-existent. Strange.

oneswitch, 2017-08-21:
D-PAD fixed. Was outputting at +1 not +100. "X" button issue on PS3 with JetPack Joyride remains. Not fully tested this original issue though.

J2Kbr, 2017-10-31:
The firmware 0.97B should now work with Raspberry Pi / RetroPie when set to use the XB360 output protocol.

oneswitch, 2017-11-10:
Connecting, but not quite working correctly. For instance, in RetroPie, you'll get a kind of bounced/double-press effect navigating the menus (making it very hard to select anything deliberately). Good progress though.

J2Kbr, 2017-12-01:
Hi, I am investigating this and need some extra info in order to progress, as the following:

1. On the Device Monitor, can you also see the inputs being double pressed?
2. What is/are the controller(s) connected to the Titan Two?
3. Did you have any script or gamepack running?

Thanks

J2Kbr, 2017-12-08:
Hello Barrie,

Regarding your previous email, the idea is very interesting and, I believe, it is feasible. If you need anything from me, please let me know.

Thank you for the information below. Luckily the RetroPie is an open source project, so I can check the code to verify how the Xbox 360 is handled, specially for the initialization part. It seems it have at least two drivers for the Xbox 360 controller, as the following:

xpad: https://github.com/paroj/xpad
-and-
xboxdrv: https://github.com/zerojay/xboxdrv

Could you please let me know which one you are using with your RetroPie?

Thanks,

Jefferson

From: Barrie Ellis
Sent: Friday, December 8, 2017 09:55
To: J2Kbr/TitanTwo
Cc: Jefferson Koppe; State change
Subject: Re: [J2Kbr/TitanTwo] Raspberry Pi / RetroPie: (in Xbox 360 outputprotocol) worked with T1 but won't with T2 (#54)

Hi Jefferson.

This is what I've tried....

1. Latest firmware (8/12/2017)
2. Forced Xbox 360 protocol.
3. With power and cable to PC, I can see the controller responding as it
should.
4. When I connect the Output to the Raspberry Pi, the Monitor no longer
responds, so I can't check if a double-press is being seen by the monitor.
5. The Raspberry Pi responds to the controller but still exhibits the
double-press / bounced controls.

The connected Xbox 360 (Microsoft official controller - black one) never
settles on a fixed LED. It constantly flashes.

The descriptor is as follows...

[USB DESCRIPTOR]
12 01 00 02 FF FF FF 08 5E 04 8E 02 14 01 01 02 03 01 09 02 99 00 04 01 00
A0 FA 09 04 00 00 02 FF 5D 01 00 11 21 00 01 01 25 81 14 00 00 00 00 13 01
08 00 00 07 05 81 03 20 00 04 07 05 01 03 20 00 08 09 04 01 00 04 FF 5D 03
00 1B 21 00 01 01 01 82 40 01 02 20 16 83 00 00 00 00 00 00 16 03 00 00 00
00 00 00 07 05 82 03 20 00 02 07 05 02 03 20 00 04 07 05 83 03 20 00 40 07
05 03 03 20 00 10 09 04 02 00 01 FF 5D 02 00 09 21 00 01 01 22 84 07 00 07
05 84 03 20 00 10 09 04 03 00 00 FF FD 13 04 06 41 00 01 01 03

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 2E
2E 03 A9 00 4D 00 69 00 63 00 72 00 6F 00 73 00 6F 00 66 00 74 00 20 00 43
00 6F 00 72 00 70 00 6F 00 72 00 61 00 74 00 69 00 6F 00 6E 00

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 16
16 03 43 00 6F 00 6E 00 74 00 72 00 6F 00 6C 00 6C 00 65 00 72 00

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 10
10 03 31 00 34 00 31 00 39 00 46 00 33 00 43 00

6. After a short-while I got constant disconnect / reconnect issues.
Possibly due to using extension cables? The Titan Two became hot. Something
crashed the RetroPie (never seen that before).

Hope this is helpful in some way.

Barrie

oneswitch, 2017-12-08:
I'm using the xboxdrv. I managed to get it into a Playstation title (Music
2000) where you can better see what the controls are doing. It's impossible
presently to hold a button it seems. For instance, the shape buttons can be
triggered, but only for a tiny fraction of second. As if the controls are
just very briefly pulsed on then off, even if you hold them.

Good luck!

Barrie

J2Kbr, 2017-12-08:
Hi Barrie,

Thank you for confirming the driver is xboxdrv.

I am wondering if the issue is related with the “full passthru” mode of the Titan Two, where it makes a virtual connection between the host (RetroPie) and the controller if the protocol is the same. Could you please test again using another controller on the Titan Two? With the output protocol still set to XB360. The controller can be from PS3 or PS4.

Jefferson

From: Barrie Ellis
Sent: Friday, December 8, 2017 12:05
To: J2Kbr/TitanTwo
Cc: Jefferson Koppe; State change
Subject: Re: [J2Kbr/TitanTwo] Raspberry Pi / RetroPie: (in Xbox 360 outputprotocol) worked with T1 but won't with T2 (#54)

I'm using the xboxdrv. I managed to get it into a Playstation title (Music
2000) where you can better see what the controls are doing. It's impossible
presently to hold a button it seems. For instance, the shape buttons can be
triggered, but only for a tiny fraction of second. As if the controls are
just very briefly pulsed on then off, even if you hold them.

Good luck!

Barrie

oneswitch, 2017-12-08:
Interestingly, using a PS4 controller in INPUT-A, it works perfectly.
Perhaps it's just an Xbox 360 controller that will cause problems. I do use
quite a few Xbox 360 controllers still in effect (e.g. PS2 controllers
through a PS2 to Xbox 360 USB adapter), so this would be really handy to
fix.

Cheers,

Barrie

J2Kbr, 2017-12-08:
I am looking at the xboxdrv source code and noticed it support for the XB360 chatpad and headset.

I believe is the chatpad support what is causing problems when the Titan Two is full passthru mode, specially after reading this note:

Chatpad WARNING: This is preliminary code, it will crash your gamepad when xboxdrv is started multiple times and won't provide proper keymapping for any of the umlauts and special characters.

I am not familiar with RetroPIE, I wonder if there is any file or configuration interface to disable the chatpad support. By the xboxdrv documentation the chatpad is enabled/disabled via command line parameter.

Jefferson

From: Barrie Ellis
Sent: Friday, December 8, 2017 12:55
To: J2Kbr/TitanTwo
Cc: Jefferson Koppe; State change
Subject: Re: [J2Kbr/TitanTwo] Raspberry Pi / RetroPie: (in Xbox 360 outputprotocol) worked with T1 but won't with T2 (#54)

Interestingly, using a PS4 controller in INPUT-A, it works perfectly.
Perhaps it's just an Xbox 360 controller that will cause problems. I do use
quite a few Xbox 360 controllers still in effect (e.g. PS2 controllers
through a PS2 to Xbox 360 USB adapter), so this would be really handy to
fix.

Cheers,

Barrie

oneswitch, 2017-12-08:
Have a little extra info here:
https://retropie.org.uk/forum/topic/10380/switch-to-a-new-fork-of-xboxdrv-in-order-to-get-new-features-such-as-chatpad-support

I'll try connecting via a forced PS3 mode, although I'm fairly sure this
won't be the ideal solution.

Cheers,

Barrie

oneswitch, 2017-12-08:
> Have a little extra info here: https://retropie.org.uk/
> forum/topic/10380/switch-to-a-new-fork-of-xboxdrv-in-order-
> to-get-new-features-such-as-chatpad-support
>
> I'll try connecting via a forced PS3 mode, although I'm fairly sure this
> won't be the ideal solution.
>
> Cheers,
>
> Barrie

I tried the following (BTW, there's no scripts running)....

1. PS3 forced mode.... Wasn't recognised by RetroPie at all in my testing.
2. Xbox 360 forced mode - works perfectly with a PS4 controller attached.
3. Xbox 360 forced mode - works perfectly with a Nintendo Switch controller
attached (the Pro controller).
4. Xbox 360 forced mode - with 2 different official Xbox 360 controllers,
and also a One Hand Controller via a PS2 to Xbox 360 adapter - all
misbehaving.

I wonder if a Raspberry Pi mode could be added to the Output Protocols,
that does away with passing through the communication side of things if an
Xbox 360 controller is attached? Might be the solution.

RetroPie is hugely popular, and would be a great way for a lot of disabled
players to get access to previously inaccessible games consoles:
https://retropie.org.uk/

Fingers crossed. :)

J2Kbr, 2017-12-08:
Hello Barrie,

Thank you for reporting all results with different controllers and configurations.

I found a interesting page that may help us, here:

https://github.com/RetroPie/RetroPie-Setup/wiki/Xbox-360-Controller

First is important to note this page also states issues with xboxdrv. I wonder if you can try with the xpad driver to check if it solves the problem, the linked page also explains how to enable xpad, its seems relatively simple. Maybe it will be necessary uninstall xboxdrv first, but not entirely sure.

If xpad don’t solves the problem, please check the command parameters of xboxdrv at /etc/rc.local

You should find something like this:

xboxdrv --trigger-as-button --wid 0 --led 1 --deadzone 4000 –silent

Check if there is any reference to chatpad on the parameters, if yes, remove.

Looking forward for your results.

Jefferson

From: Barrie Ellis
Sent: Friday, December 8, 2017 14:25
To: J2Kbr/TitanTwo
Cc: Jefferson Koppe; State change
Subject: Re: [J2Kbr/TitanTwo] Raspberry Pi / RetroPie: (in Xbox 360 outputprotocol) worked with T1 but won't with T2 (#54)

I tried the following (BTW, there's no scripts running)....

1. PS3 forced mode.... Wasn't recognised by RetroPie at all in my testing.
2. Xbox 360 forced mode - works perfectly with a PS4 controller attached.
3. Xbox 360 forced mode - works perfectly with a Nintendo Switch controller
attached (the Pro controller).
4. Xbox 360 forced mode - with 2 different official Xbox 360 controllers,
and also a One Hand Controller via a PS2 to Xbox 360 adapter - all
misbehaving.

I wonder if a Raspberry Pi mode could be added to the Output Protocols,
that does away with passing through the communication side of things if an
Xbox 360 controller is attached? Might be the solution.

RetroPie is hugely popular, and would be a great way for a lot of disabled
players to get access to previously inaccessible games consoles:
https://retropie.org.uk/

Fingers crossed. :)

oneswitch, 2017-12-08:
> issuecomment-348425471

I'll try to do this ASAP. Thanks as ever for helping.

Ideally the fix will ultimately be within the T2 though, as a lot of users
would struggle with messing about with the RetroPie settings.

Cheers,

Barrie

oneswitch, 2017-12-08:
> 350258724>,

Removing the xboxdrv and replacing with xpad makes little positive
difference, if any, it seems.

I can't configure the Xbox controller in RetroPie (when connected via the
T2) as you have to hold a button down for about a second. Controls are
being passed through, but only pulsed for a fraction of a second, as before.

There's no extra options for the xpad driver (unlike the xboxdrv) that I
can find.

I couldn't find the rc.local file you spoke of, using FileZilla.

J2Kbr, 2017-12-12:
Good news, I was able to reproduce this issue using a Linux VM and xboxdrv. The fix should be out yet today, firmware revision 0.98B-1

J2Kbr, 2017-12-12:
Hello Barrie,

Good news, I was able to reproduce this issue using a Linux VM and xboxdrv.
The fix should be out yet today, firmware revision 0.98B-1

Thank you for all your help with this.

Jefferson

From: Barrie Ellis
Sent: Friday, December 8, 2017 16:17
To: J2Kbr/TitanTwo
Cc: Jefferson Koppe; State change
Subject: Re: [J2Kbr/TitanTwo] Raspberry Pi / RetroPie: (in Xbox 360 outputprotocol) worked with T1 but won't with T2 (#54)

Removing the xboxdrv and replacing with xpad makes little positive
difference, if any, it seems.

I can't configure the Xbox controller in RetroPie (when connected via the
T2) as you have to hold a button down for about a second. Controls are
being passed through, but only pulsed for a fraction of a second, as before.

There's no extra options for the xpad driver (unlike the xboxdrv) that I
can find.

I couldn't find the rc.local file you spoke of, using FileZilla.

closed 2017-12-12 by J2Kbr

oneswitch, 2017-12-12:
> issuecomment-350279121

Brilliant, Jefferson!

Seems to work very well with the xpad driver too, which might be the one to
go for. Didn't work well at all until this fix. Top job!

Barrie