J2Kbr/TitanTwo #55: Nintendo Switch compatibility

oneswitch, 2017-06-14:
Would be amazing to get this featured. I see there are some third party controllers.

http://www.8bitdo.com/snes30-sfc30/

oneswitch, 2017-06-20:
8bitdo say with their latest firmware update, their (lacking analogue) controller will work with Mario Kart 8 and Zelda. No motion control though.

colin1993, 2017-06-20:
I would love the Titan Two to be compatible with the Nintendo Switch so that I can use Joy2Key and UCR thanks to Barrie Ellis and Special Effect. I have never been able to access any Nintendo console due to the controllers, and it is a shame because they make great games that I'd really love to play! Please help!

oneswitch, 2017-07-04:
Looks possible. Brook have just posted a video with PS3/PS4 to Switch compatibility with their basic adapters: https://www.youtube.com/watch?v=2PC3JLs34Q4

oneswitch, 2017-07-04:
More info here: https://www.reddit.com/r/WiiUHacks/comments/5j5t2m/xbox_one_controller_working_on_wii_u/

netarchy, 2017-07-25:
Nintendo Switch support would be good, even better if we could additionally map the gyro to the mouse as an option instead of the right thumbstick, with possibly support for a toggle between stick/gyro. Ideal for any games that support motion controls.

oneswitch, 2017-07-26:
Seems that you can Daisy Chain a Titan One into a Brook PS3/PS4 to Wii-U (latest version) with the latest Switch firmware update from their firm. Not sure if the Daisy-Chain effect is causing some lagging. 

Colin McDonnell proved it works on a Nintendo Wii-U. I've had a few minutes with it on a Switch today. Looks hopeful. :)

J2Kbr, 2017-09-16:
Support for Nintendo Switch and Pro controller added to Titan Two firmware 0.95B

closed 2017-09-16 by J2Kbr