J2Kbr/TitanTwo #60: Add support for Arduina Leonard as Keyboard

J2Kbr, 2017-06-28:
```
[USB DESCRIPTOR]
12 01 00 02 EF 02 01 40 41 23 36 80 00 01 01 02 03 01 09 02 64 00
03 01 00 A0 FA 08 0B 00 02 02 02 01 00 09 04 00 00 01 02 02 00 00
05 24 00 10 01 05 24 01 01 01 04 24 02 06 05 24 06 00 01 07 05 81
03 10 00 40 09 04 01 00 02 0A 00 00 00 07 05 02 02 40 00 00 07 05
83 02 40 00 00 09 04 02 00 01 03 00 00 00 09 21 01 01 00 01 22 2F
00 07 05 84 03 40 00 01

[STRING 1 DESCRIPTOR]
00 00 00 03 00 00 02 01 80 06 03 01 04 09 00 18
18 03 41 00 72 00 64 00 75 00 69 00 6E 00 6F 00 20 00 4C 00 4C 00
43 00
```

cr33dk1ller, 2017-06-29:
not sure if it matters, but after the recent update, i did the fetch descriptors again for the leonardo keyboard and the first line from the string 1 changed from 00 00 00 03 to 00 00 00 02, the rest remains the same. to be perfectly honest tho, there is a possibility that the original descriptor was 02 and i just transcribed it incorrectly. i had to transcribe it from a screenshot because of the disconnecting problems ive been having (something to do with my computer and not the t2). i hope j2kbr you can check the leonardo keyboard descriptors yourself and confirm them before implementing them into the firmware update just to be on the safe side.

J2Kbr, 2017-07-05:
Hi cr33dk1ller. Unfortunately I couldn't decode the descriptors you sent. It has errors and seems to be incomplete, for example, it must have a section called `[INTERFACE <no> HID DESCRIPTOR]`.

In any case, I did many improvements on the firmware related to the support of keyboards. I expected the Leonardo to work after this update. Please reopen this issue if not. thanks.

closed 2017-07-05 by J2Kbr

cr33dk1ller, 2017-07-06:
thanks for your reply J2Kbr. The reason the descriptors was incomplete was because of the USB issue I have been facing with my primary computer. I managed to fetch the descriptors on a second computer and this time I hope it is complete. I understand your update should be out soon and the leonardo should already work, but incase you need it, here it is.

[USB DESCRIPTOR]
12 01 00 02 EF 02 01 40 41 23 36 80 00 01 01 02 03 01 09 02 64 00 03 01 00 A0 FA 08 0B 00 02 02 02 01 00 09 04 00 00 01 02 02 00 00 05 24 00 10 01 05 24 01 01 01 04 24 02 06 05 24 06 00 01 07 05 81 03 10 00 40 09 04 01 00 02 0A 00 00 00 07 05 02 02 40 00 00 07 05 83 02 40 00 00 09 04 02 00 01 03 00 00 00 09 21 01 01 00 01 22 2F 00 07 05 84 03 40 00 01 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 18 
18 03 41 00 72 00 64 00 75 00 69 00 6E 00 6F 00 20 00 4C 00 4C 00 43 00 

[STRING 2 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 02 04 09 00 22 
22 03 41 00 72 00 64 00 75 00 69 00 6E 00 6F 00 20 00 4C 00 65 00 6F 00 6E 00 61 00 72 00 64 00 6F 00 

[STRING 3 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 03 04 09 00 0C 
0C 03 48 00 49 00 44 00 50 00 43 00 

[INTERFACE 2 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 02 00 2F 
05 01 09 06 A1 01 85 02 05 07 19 E0 29 E7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 03 95 06 75 08 15 00 25 65 05 07 19 00 29 65 81 00 C0

cr33dk1ller, 2017-07-06:
hey J2Kbr, just wanted to let you know the arduino leonardo keyboard is now recognized. xD

J2Kbr, 2017-07-06:
That is awesome!! :) Thanks for the confirmation.

sky418, 2017-07-06:
Logitech G 13
[USB DESCRIPTOR]
12 01 00 02 00 00 00 08 6D 04 1C C2 03 01 00 01 00 01 09 02 29 00 01 01 00 80 FA 09 04 00 00 02 03 00 00 00 09 21 11 01 00 01 22 3D 00 07 05 81 03 08 00 02 07 05 02 03 40 00 01 

[STRING 1 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 01 04 09 00 08 
08 03 47 00 31 00 33 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 3D 
06 00 FF 09 00 A1 01 15 00 26 FF 00 09 01 85 01 95 07 75 08 81 02 85 03 09 02 96 DF 03 91 02 85 07 09 03 95 04 B1 02 85 04 09 04 B1 02 85 05 09 05 B1 02 85 06 09 06 96 01 01 B1 02 C0