J2Kbr/TitanTwo #61: Titan Two Device API.

paname75, 2017-07-06:
This is a Feature request for an API for software development. Similar to T1 device API, the API would allows programmers to develop plugins and applications to interact directly with the Titan Two device.

Ideally this API would provide a 64bit DLL.

J2Kbr, 2017-08-10:
The API will be implemented, hopefully, yet this year. 32 and 64 bits, as well as Mac/Linux variants will be available.

The idea is connect to the Titan Two as a "Virtual HID Device", such as keyboard, mouse, joystick or any combination of these.

bootstraponline, 2017-12-16:
Is there any update on an API? one would be amazing.

J2Kbr, 2018-05-22:
1. Ability to set the controller output protocol. Very useful to force the unit into Xbox 360 mode if being used on a PC, Raspberry Pi, Android device. Will prevent needing to keep resetting in GTuner IV.

2. Ability to turn off FFB / rumble motors.

3. Colour LED access. Perhaps the 7-segment display too.

4. Full access to controls, including touch-pad and six-axis.

5. Keep "ACCLEL B" clear as used for communication with the Game Controller Mixer?? Perhaps I should speak to Graham about this too.

oneswitch, 2018-12-21:
Still hoping for the API, so that the likes of Universal Controller Remapper can interface directly with the brilliant Titan Two (as is possible with the Titan One).

evilC, 2018-12-21:
> The idea is connect to the Titan Two as a "Virtual HID Device".  
I assume you are talking about sending to the T2 here, rather than reading from?  
This sounds a lot more complex for the consumer than hitting a dedicated API

willwade, 2019-08-12:
I realise this maybe outside the scope of this project.. But I'm throwing it out there anyway.. I would personally like something like a RPC server and an API to call that - so can call the titantwo from other devices and RPC is not linked to a particular language (i.e. wouldnt need C bindings). Use case: User on iOS with a custom interface or switch scanning; selects an action (X, Y, Joy left) - and this then calls the titantwo wirelessly (Building a BT Gatt Interface could be another option but that's not straightforward). 

NB: I'm kinda stealing this idea from our own RelayKeys project (https://github.com/AceCentre/RelayKeys)  - which does something similar. It's ok - but the titantwo could allow us to do more than BT HID.

J2Kbr, 2020-03-03:
With 1.07 update (Gtuner IV and Titan Two firmware), we now have a fully functional Python scriptable interface to communicate with the Titan Two device, with ability of read controller inputs and send commands to a running GPC script on the Titan Two.

This interface was originally made for Computer Vision, however is possible to create/run Python scripts for any purpose.

So I am wondering if the API is still needed, as you guys could easily create "programs" and communicate with the Titan Two using Python. Please let me know your thoughts.

willwade, 2020-03-03:
That sounds good to me - do you have a pointer for the docs?

J2Kbr, 2020-03-04:
Thank you willwade for your feedback.

This feature still in beta period, you can find more details here:

https://www.consoletuner.com/forum/viewtopic.php?f=2&t=14717

cgalway-iflo, 2020-03-04:
@J2Kbr I would much rather a C/C++/.NET API thanks :(
Implementing support for Python into our .NET app just for this is kind of undesirable

HeyItsLollie, 2020-03-05:
As someone who knows nothing about Python, it's hard to fully judge this without proper documentation — hopefully that's being focused on. It sounds like a cool step forward, but I am still leaning towards an API.

kmcnaught, 2020-04-23:
I'd really like an API too, ideally C/C++/.NET - my use case would be adding the Titan Two as an optional output to a voice control app so that users don't have to launch gTuner and KMG capture to be able to hook up a voice command to a consoler output via the Titan. From the limited information about the python bindings, it looks like they run inside gTuner? So am I right to think you couldn't have a standalone python executable sending control signals to the T2?

xXSkyWalkerXx1, 2020-05-05:
> I'd really like an API too, ideally C/C++/.NET - my use case would be adding the Titan Two as an optional output to a voice control app so that users don't have to launch gTuner and KMG capture to [...]

That's right.

francis-specialeffect, 2020-05-14:
Having the python bindings is a great addition, but I too agree that creating an API should still be considered for supporting a wider range of use cases. The instances mentioned of having to add support for python in programs, and sending commands to the T2 without GTuner needing to be open both seem like valid reasons to keep considering API support.

evilC, 2020-05-14:
@francis-specialeffect From what I understand in this thread (See response by xXSkyWalkerXx1 above), adding python support to an app is not even an option, as the python support is only for scripts running inside GTuner