J2Kbr/TitanTwo #69: any using "while" make's T2 crash

sky418, 2017-08-29:
Any using "while" function makes T2 crash

J2Kbr, 2017-09-03:
Please include an example of `while` code that crashes the Titan Two. I tested with the script below and it worked as expected (without crash).
```
main {
    int i = 20;
    while(i) {
        --i;
    }
}
```

J2Kbr, 2017-09-16:
Accordingly the PMs we exchanged on the forums, I am considering this as solved. If you still have any doubts, please PM me.

closed 2017-09-16 by J2Kbr

sky418, 2017-09-18:
> Jefferson Koppe <mailto:notifications@github.com>
> 3 сентября 2017 г., 17:07

Sorry for long time answer
while making script was made a mistake and Compile report - OK no errors
this one
################
int s;
int r;
init {
    s = 0;
}
main {
  if(event_active(BUTTON_10)){
     test(BUTTON_10);
     }
}
void test (uint8 active){
     s ++;
     while ( s < 10){
     r = s;
     }
     if(event_active(active)) {
     printf("R = %d",r);
     }

}
##########
And T2 crash.
Then correct to
################
int s;
int r;
init {
    s = 0;
}
main {
  if(event_active(BUTTON_10)){
     test(BUTTON_10);
     }
}
void test (uint8 active){
     while ( s < 10){
     s++;
     r = s;
     }
     if(event_active(active)) {
     printf("R = %d",r);
     }

}