J2Kbr/TitanTwo #70: LED Status from PS4 console "leaking" to controller

J2Kbr, 2017-09-03:
**paname75**
> I should add that even if there is just a DS4 controller on the T2 (no xim4) and have a simple script which set T2 led color to red like this one , when the destiny super bar is full and console send pulsating yellow led change, the T2 led stay red per script but blink very fast (almost not noticeable) the device led setting is set to gamepack/script and console led messages should be ignored but somehow it isnt internally in the T2 and it transfert the fast blink to controllers, that's why it keep sending update to connected controller which overflow the slow xim4 queue when ds3 is used.
> 
> i think there is something odd in the report filtering and this is 100% fully reproductible in destiny just wait for the super bar to fill up (3-4 min in patrol or just launch a mayhem rumble private match bar will fill in 30s)
> 
> Also the intro from BlackOps 3 do the exact same E4 crash with xim4+DS3 but the fast blinking isn't visible like in destiny super bar full.
> 
> ```
> #pragma METAINFO("ledtest", 1, 0, "paname")
>  
> char CurLedColor   = 'N';
> char PrevLedColor  = CurLedColor;
>  
> init {
>  
>   ColorLED('R');
>  
> }
> main {
>  
> }
> void ColorLED(char Color) { 
> 	fix32 Color1, Color2, Color3, Color4;
>  
> 	if (CurLedColor == Color) return;
> 	PrevLedColor  = CurLedColor;
>  
> 	if(Color == 'O'){Color1 = 0.00;    Color2 = 0.00;    Color3 = 0.00;    Color4 = 0.00; }
> 	if(Color == 'B'){Color1 = 50.0;    Color2 = 0.00;    Color3 = 0.00;    Color4 = 0.00; }
> 	if(Color == 'R'){Color1 = 0.00;    Color2 = 50.0;    Color3 = 0.00;    Color4 = 0.00; }
> 	if(Color == 'G'){Color1 = 0.00;    Color2 = 0.00;    Color3 = 50.0;    Color4 = 0.00; }
> 	if(Color == 'P'){Color1 = 0.00;    Color2 = 0.00;    Color3 = 0.00;    Color4 = 50.0; }
> 	if(Color == 'C'){Color1 = 50.0;    Color2 = 0.00;    Color3 = 50.0;    Color4 = 0.00; }
> 	if(Color == 'A'){Color1 = 0.00;    Color2 = 50.0;    Color3 = 50.0;    Color4 = 0.00; }
> 	if(Color == 'W'){Color1 = 50.0;    Color2 = 50.0;    Color3 = 50.0;    Color4 = 50.0; }
>  
>     CurLedColor = Color;
> 	led_set(LED_1, Color1, 0);
> 	led_set(LED_2, Color2, 0);
> 	led_set(LED_3, Color3, 0);
> 	led_set(LED_4, Color4, 0);
>     return;
>  
> }
> ```
 **paname75**

> Hi J2Kbr,
> 
> Do this also fix the fast blinking LED when in destiny the super is full and that even if the LED config is Gamepack/script and just with a DS4 on port A (nothing else)
> 
> it acts as if there is a "leak" of the LED message from output port to the other port when it shouldn't.
> if you set the color to red using the provided script. You will see the blinking no matter what when the super bar is full in destiny PS4.
> 
> you can test this without any device but PS4 and DS4 on the titan2.

paname75, 2017-09-03:
the leaking i'm talking about is mainly observable on the T2 LED and occurs when super is full and stop when super is used (R1+L2 in the game).  that's because the game send a very fast pulsating led changes when super is full.

i really think that's a major reason for random E4 people see with various "old" device because it is tied to game condition.

J2Kbr, 2017-10-03:
Issue fixed on the next revision of the firmware 0.95B (to be released soon).

closed 2017-10-03 by J2Kbr

paname75, 2017-10-05:
Hi,
I still see the led fast blink on the T2 led itself so maybe just an artifact.