J2Kbr/TitanTwo #71: Controller "locked" with Wireless receiver

lemanr, 2017-09-08:
i have a problem playing FFXV in ps4

The config

Ps4 --> Titan Two --> Xbox One Wireless Receiver --> Xbox one Elite Controller

Many times, the Controller is blocked, and doing the last action received, for example, if i an doing an attack pressing B, it continues attacking without end, the solution is disconnect the Xbox one Wireless Receiver and reconnect it

When the problems happend, the Xbox one Elite controller still paired with the receiver (the status led is power on)

Sorry for me english

J2Kbr, 2017-10-03:
I experienced this -_or very similar to this_- during this past week. The issue with me was not related with the Wireless Receiver, but with the authentication of the Titan two with the PS4 console. Please confirm if in the setup described you also had a PS4 controller connected to the Titan Two, or not. Thanks.

lemanr, 2017-10-04:
Yes, i have the ps4 controller connected to the titan two

i will try again to check if whit the last firmware its persist

Thank you for all your work

lemanr, 2017-10-18:
Hello Again

I try it with the last firmware, and i continue with the error, i try playing with destiny 2

Its recommended to connect the titan two to an specific usb port of the PS4?

Thanks for all

DebuRanger, 2017-10-18:
HI,
I'm new here and also have the same issue.  I'm on 0.95.1.1
I have the PS4 controller on port 2.
I have the SD card in the T2 and have the firmware on it as per Gtuner's automatic setup.

What I think is happening is that the connection is getting dropped.  The little white light on the XboxOne Wireless adapter turns off after a bit and then the connection is broken.  I believe what ever the last input keeps getting sent by the T2 to the PS4 and thus you see the spinning or jumping or whatever after the connection is lost.

I can use the XboxOne Elite with a wired connection fine.  So I feel the authentication is fine.  It has something to do with the dongle and how it might be talking with the micro-controller.

DebuRanger, 2017-10-25:
J2Kbr,

I tried with the firmware released this week.  I believe it was 0.96.x and the bug is still there.

J2Kbr, 2017-10-31:
**ChaosRocket**
MS wireless dongle (1)
> XB1 and using black MS wireless dongle.
> 
> Buttons A and B are swapped. Latest firmware and software.
> 
> No translation tables, T2 reads “0” on screen, output is selected as XB1
> 
> https://drive.google.com/open?id=0B-TDPJl60rmEcndqWFAtWEJxWlE
> 
> Regards - Andy

**ChaosRocket**
MS wireless dongle (2)
> On XB1 with black MS wireless dongle with elite controller
> 
> Controller disconnects after 3 min controller xbox button flashes @3.30 controller switches off.
> 
> Turning controller back on, doesn’t reconnect.
> 
> Dongle pairing button doesn’t work until re-insertion.

> Special note re-insertion does cause automatic re-pair when controller is switched on.

> Further note controller always switches off after 3.30 even if using.

J2Kbr, 2017-10-31:
Hi Guys, I am trying reproduce this issue without success,. Sorry for the fix taking so long. I need few more info, as the following:

1. Do you also have any accessory connected to the XB1 controller? such as chatpad or headsets?
2. The issue also happens if the Titan Two in only connected to the console? (not connected at the same time on PC/Gtuner).
3. The issue only happens when playing a game? (which game?). _Here I want to know if the problem also happens on the console system menu, without a game running_.
4. If having the Titan Two only connected to PC, only with the wireless receiver and XB1 controller synced. With Gtuner IV opened on Device Monitor to check the controller inputs, the issue also happens?

lemanr, 2017-10-31:
Hello J2Kbr

I answer you

1. I havent any accesory connected, only the controller
2. The Titan Two is only connected to the PS4
3. In my case, it happens when i play, but because its the only use of my ps4, when i power on it is only for play, i will try in the menus, About the games, it happens to my in FFXV & destiny 2, i will try with other games like GT Sport
4. Never happens to me, i used Gtuner many time to configure it, re-mapping, etc,etc, and never has the issue

Again thank for your work and sorry my english

DebuRanger, 2017-11-05:
Hi,

This is the version of the adapter I am using.  There is a newer one that is smaller.  I have the chunky one.
https://www.groupon.com/deals/gg-microsoft-xbox-wireless-adapter-for-windows

As of firmware 0.97.1.0 It stayed connected.  
Reading the change log I don't see anything there.

I'm going to do more testing this week and see if it stays solid.

J2Kbr, 2017-12-01:
According user's reports, issues with the Xbox Wireless Adapter seems to be solved after the firmware 0.97B-2 release. Please re-open this GitHub issue if you are still having problems. Thanks.

closed 2017-12-01 by J2Kbr