J2Kbr/TitanTwo #72: Nintento Switch Multiple Titan Two's

francis-specialeffect, 2017-09-27:
Great work getting Nintendo Switch support for the Titan Two. I did however encounter a problem when using two Titan Two's together - they were both recognised as the same controller, as opposed to two separate controllers.

J2Kbr, 2017-10-03:
This happens because the Titan Two is using a hard coded "ID". One solution could be use part of the Titan Two's serial number to generate an unique id for each device. I will see if this is feasible, as this ID is in fact the Bluetooth Address of the device and it may be important attach it to the BD Addr of the upcoming Bluetooth Module.

J2Kbr, 2017-10-31:
With the upcoming support for wireless connection with the Nintendo Switch this limitation will be naturally overcome, as the BD Address of the Bluetooth Module will be used instead of hardcoded value.

closed 2017-10-31 by J2Kbr