J2Kbr/TitanTwo #74: Mic audio issues

J2Kbr, 2017-10-01:
Investigate user's reports of the mic audio are presenting occasional distortions - most notable on XB1, but also on PS4.

hiroshimus, 2017-10-06:
Hey J2KBR,

You know what, I've been using the secondary xbox one controller connected to Port B but it would seem that my voice keeps getting distorted from time to time. People were complaining about it a lot today, I just had to turn off my mic.

There have been times when it's been working pretty well though.

As a matter of fact I'm playing right now and I'm having the same issue.

Have you had any progress with this?

=)

J2Kbr, 2017-11-14:
Gärtner
https://www.consoletuner.com/forum/viewtopic.php?p=54446#p54446
> Hello, I would like to report here because I have exactly the same problem with cable headset. New PS4 Controller in USB-B with headset and old PS4 controller in USB-A -> voice is slightly distorted/scratchy. Only new Controller in USB-A with headset, all ok. Maybe this will help you.

J2Kbr, 2017-12-07:
Firmware revision 0.97B-8 fixes mic audio issues by requesting isochronous data earlier than to what is specified by bInterval.

closed 2017-12-07 by J2Kbr