J2Kbr/TitanTwo #76: (Feature Request) Have The Friendly Name Stored on the T2, not computer.

MaxRat08, 2017-10-08:
When working with multiple devices across multiple computers, it is easy to get them mixed up. Tt would be useful if the titan 2 would save its nickname internally so that when plugged into new computer it automatically comes up with that name instead of its serial number.

prototypeGtuner, 2017-10-16:
You can have a Titan Two folder in a flash drive and run Gtuner from there to have the same settings. Or you can also use dropbox to save your settings online.

J2Kbr, 2017-10-31:
Save the friendly name on the Titan Two internal flash memory at this point is not possible, as all space is allocated for specific ends. It would be possible to save the friendly name on the sd card, but this would require the sd card for this purpose, not allowing users without it set friendly name. As the configuration of the friendly name is an one time procedure, the solution is set it on each computer.

closed 2017-10-31 by J2Kbr