J2Kbr/TitanTwo #84: McCthulu PCB Fight Stick Board Double Moves on XB1 Menu

DebuRanger, 2017-11-05:
Hi,

So when I navigate the XB1 menu using the McCthulu fight stick, it moves 2x for each move the the joystick.  So for example, if I move down one tile, it moves down two tiles.

The T2 sees it fine and it seems to work ok in the game mode like on Killer Instinct a fighting game.

When looking into Gtuner (I still don't know how to map anything...sigh) I can go to the Device Manager and see the buttons all being pressed and joystick registers.  

If you would like a webex or something to see the output or what happens on screen, let me know.  I'd be happy to assist.

J2Kbr, 2017-12-01:
Hi, Please give more details, as the following. 

1. On the Device Monitor, can you also see the inputs being double pressed?
2. It the McCthulu the only de vice connected to the Titan Two?
3. Did you have any script or gamepack running?

Thanks.

J2Kbr, 2017-12-12:
In addition to the above questions, please also let me know:

4. On the Device Monitor at the center area, the USB port the controller is connected and USB C shows "Full Passthru"? or just "XB1 [IN]" and "XB1 [OUT]" respectively?

Thanks.

PablosScripts, 2018-01-04:
I had this issue with my fight stick, the solution is here: 

https://www.consoletuner.com/forum/viewtopic.php?f=27&t=5676

DebuRanger, 2018-01-05:
I'm so sorry for not following up with everyone.  I got totally side tracked.
I'll check the script out.  Thank You.

J2Kbr, 2018-01-25:
Assuming this issue as solved, if not, Please comment. Thanks.

closed 2018-01-25 by J2Kbr