J2Kbr/TitanTwo #87: Nintendo Switch Pro Controller no longer working on Xbox One.

MaxRat08, 2017-11-14:
In reference to my post on the forum ([https://www.consoletuner.com/forum/viewtopic.php?f=27&t=7295](url)). 
I will update this with new info as I am able to test more scenarios.

J2Kbr, 2017-11-24:
I couldn't reproduce this issue. The test I did was:
1) Connect the Titan Two to Xbox One (first model) with Elite controller on INPUT-B. After the authentication I plugged the Switch Pro controller on INPUT-A, everything worked as expected. I am now just waiting a game update to install to test while playing, but I expect it to work as well.

If your procedure was different than mine or if you have any extra info on this issue, please let me know.

Thanks.

MaxRat08, 2017-11-25:
I have been unable to get this to repeat as of the latest software update. May have been related to the brown out voltage change? It is working now though.

J2Kbr, 2017-11-25:
You are right, possible this is related with the brown-out voltage, which was fixed in a recent update. So let's close this issue. Thanks.

closed 2017-11-25 by J2Kbr