J2Kbr/TitanTwo #88: wireless adapter for windows not powering on

IMulligaNI, 2017-11-17:
Just received my titan two today but have encountered a problem, I'm trying to use a Xbox one controller on my Ps4 pro through a wireless adapter but the white led is not showing on the adapter as if it isn't powered on. when using Gtuner it states that usb 'a' is a wireless adapter but also refuses to work. My wireless adapter works fine when plugged strait in to my computer without the titan two, so it must be issue with my device.

J2Kbr, 2017-12-01:
According user's reports, issues with the Xbox Wireless Adapter seems to be solved after the firmware 0.97B-2 release. Please re-open this issue if you are still having problems. Thanks.

IMulligaNI, 2017-12-01:
still getting this problem even after installing the latest firmware

J2Kbr, 2017-12-02:
Closing. Issue #89 replaces this.

closed 2017-12-02 by J2Kbr