J2Kbr/TitanTwo #89: Wireless adapter for Windows not Syncing to controller when using TitanTwo on PS4

IMulligaNI, 2017-12-01:
#88 
Having some issues on my new TitanTwo. I'm unable to sync my wireless adapter to a Xbox one controller when the TitanTwo is connected to my PS4 Pro, When the device is connected to Gtuner. I am able to sync the controller to the adapter and can see the emulation on both controllers, when the device is removed from computer and plugged into output to PS4. The wireless adapter doesn't sync to the controller it seems, As the white light doesn't stay active. on the Dongle

Some more info on my setup in case I missed something

- Input A - Wireless Adapter
- Input B - PS4 Controller
- All three lights show green on the TitanTwo
- Output - USB on the back of my PS4 pro
- Latest Firmware
- Micro SD formatted and configured
- No scripts active or installed on Micro SD card
- USB only mode active on PS4
- Works fine when using a wired Xbox controller on Input A
- Controller
- All three lights show green on the TitanTwo
- Output - USB on the back of my PS4 pro
- Latest Firmware
- Micro SD formatted and configured
- No scripts active or installed on Micro SD card
- USB only mode active on PS4
- Works fine when using a wired Xbox controller on Input A

#89 unable to find option to re-open my issue on #88, Still getting the same issue after updating software for Gtuner and the Firmware on my Titan2, starting to think I have Dud unit or less I haven't set something up right?

IMulligaNI, 2017-12-04:
is there something I'm not doing correctly here or is it only me that is experiencing this issue? if so how do I go by getting a replacement? As I have had my item 20 days now and it doesn't seem like its a common issue for anyone else as I have had no replies. The only reason I brought the Titan two was so I could use a wireless Xbox controller on my PlayStation if I cant do that then the item is useless to me as I already have a Cronus Max.

J2Kbr, 2017-12-12:
Hi IMulligaNI, After receiving user's reports and perform some tests it seems the XBox Wireless Adapter consume a significant amount of power. I could reproduce issues similar to what you described with the Titan Two on PS4 and having the Xbox Wireless Adapter and a DS4 connected to the Titan Two. No issues were noticed after plug the external power supply to the Titan Two.

If you have possibility, please test with an external power supply.

There is also an alternative method that does not requires the PS4 controller connected to the Titan Two and may help in this case. If you are interested, please PM me on the forums.

J2Kbr, 2017-12-21:
Relevant info from Bakiman
https://www.consoletuner.com/forum/viewtopic.php?p=55871#p55871
> I found a workaround : insert the Titan two without the wireless adapter. Wait till the display of the Titan two shows the letters u and a. Then insert the Xbox wireless adapter, two seconds later I have green light and connection. If I don’t do it like this, the light stays orange and I could wait for days to turn green. So I’m happy with the workaround, hopefully it gets easier with an update and there is no need for unplugging. My as card is 16gb, maybe this is a reason why it doesn’t work like others? Too big a card to read quickly?

bakiman, 2017-12-26:
update :  I follow the wireless setup for xbox one on ps4 (gtuner, format sd, prepare sd, insert wireless, wait till green, press connect button, connect controller). After this I put the titan two on my ps4 and it works immediately. I cannot seem to get it to work the next day. what makes it work the first time?  something preloads when connected to Gtuner? that doesnt load when on ps4 output? and stays in the memory for some time? just my thoughts, would love to help getting this to work. I will try to reformat my sd to smaller size.

J2Kbr, 2018-01-10:
More relevant info:

> I found another way to make the titan recognize the wireless adapter: first plug the Titan in via the prog port, then quickly unplug and switch to the output. Light goes green in 1 second. If you do it slow, no success.
> i don’t know why, but there is something about triggering the prog port that makes it work.

J2Kbr, 2018-01-25:
I tried reproduce the described scenarios and it worked in my end. Tested with multiple Titan Twos to be sure.

@ IMulligaNI, please check if your have this configured:

**On PlayStation 4, enable wired communication:**
`Settings → Devices → Controllers → Communication Method: Use USB Cable.`

I am closing this issue. If the problem persists, please comment to reopen. thanks.

bakiman, 2018-01-25:
I think the issue I have can be described more clearly now. It is not the controller that can’t be authenticated. It is the Xbox wireless adapter that is not recognized.

See below the most detailed description of what happens.
Thanks for reading :-)

 Xbox wireless adapter - yellow led, not green
Sent: Thu Jan 25, 2018 4:55 pm 
From: Bakiman 
To: J2Kbr 

Hi,

The wireless adapter for xbox is never recognized by the titan two when I plug in the titan in my ps4 pro. Yelled led, even after waiting 30 minutes.

The only workaround is : plug the titan two in the PlayStation via the prog port. Then pull out the usb from prog port and quickly plug into the output.

Then the led goes green after 2 seconds and I can use the device. If it doesn’t work, waiting longer does not make it recognize the wireless adapter. You have to do the prog to output trick again.

What also works is doing the initialization via gtuner as if you setup a fresh Titan two.

I bought a couple of sd cards. All have the same issue.

Did other people have this issue? Were you able to see this bug on ps4 ? Or does the Titan led turns green when starting up the ps4 with the wireless adapter in?

When it is working, it works flawless! I have had consistent perfect response! Any advice on solving this would help a lot. I don’t like to plug in plug out too much to potentially damage the device.

(Or is only my Titan two behaving a bit strange? 
Thanks for your reply.
Best regards,
Tom

J2Kbr, 2018-01-27:
@bakiman, I did many tests here, with different configurations and I believe to have partially reproduced the issue.

1. First I noticed some connection inconsistency when the output protocol is set to AUTOMATIC, configuring it to PS4 fixed this issue.

2. Without the DS4 controller on INPUT-B the Wireless Adapter initialized successfully 100% of the times (green light), However, having the DS4 controller it was verified occasional fail in the initialization. Note during this test it was used a high-quality short cable (30cm) to connect the Titan Two on PS4. With longer and/or low quality cable this issue can be more frequent.

_**PS.:** There is the possibility to use the device without the DS4 controller, if you want more details on this please send me a PM me on the forums._

closed 2018-01-27 by J2Kbr