J2Kbr/TitanTwo #91: Disconnect after choosing memory slot at SD Card

monty35, 2018-01-03:
Normal T2 behaviour at startup with Slot 0 choosen, after LB and XB Button to go the last slot at the SD card there is a regular disconnect with red light for the output. I´ve to replug the output port again to get the slot running.

System: XB1 with XB1 controller (nothing else connected)

J2Kbr, 2018-01-25:
Please post the script that is causing this issue (or confirm if the issue happens with any script).

Side note: I tried reproduce as described, here seems to be working as expected.

Memory Slots 3 and 5 with scripts.
Xbox One (no game loaded, only on the system screen),
Elite controller wirelessly;
LB+Xbox, or RB+Xbox working as expected.

Thanks.

J2Kbr, 2018-02-19:
I believe to have found the cause of this issue #113 . Fixed on Firmware 0.98B-15, please comment if the issue persists after the update. Thanks.

closed 2018-02-19 by J2Kbr