J2Kbr/TitanTwo #93: Touchpad Click and Share Button Question

Fen-Ria, 2018-01-05:
I have noticed that when using a ps4 controller that clicking the touchpad acts as the back, select and view buttons for the 360, ps3, and xbox one controllers instead of the share button like on the Titan One.

Is that intentional?

J2Kbr, 2018-01-23:
A previous firmware update changed the button assignment of the Share and Touch-click. Now the Share is BUTTON_18 and touch-click BUTTON_2, this was done to have a more consistent button mapping across the consoles.

closed 2018-01-23 by J2Kbr

Fen-Ria, 2018-01-25:
May I know why that would be considered the more consistent button mapping?

J2Kbr, 2018-01-26:
For example, in multi platform games the function of `VIEW`, `BACK` or `SELECT` buttons on PS4 is normally bidden to the `TOUCHPAD-CLICK`, and not to the `SHARE` button.

Fen-Ria, 2018-01-31:
Ah, okay.
Thank you for letting me know.