J2Kbr/TitanTwo #103: (Feature Request) Allow a separate remapper profile for each port.

MaxRat08, 2018-01-30:
Allow remapper() to have a different profile for each port.
Example:
Output: Xbox One
USB A: Switch Pro Controller -> Swap A/B and X/Y to match output.
USB B: DS4 -> Do not swap the face buttons, but swap share and touchpad click.

Can you add a way they can both be run at once?

I have tried blocking a port (set everything in the remapper to -1 so it doesn't pass a value to the console) and switching rapidly between remapper profiles (one for each port), but this uses too much memory and crashes the T2.

juroz, 2018-02-03:
Yeah i really needs this to use a joystick in combination with mouse and move controller

J2Kbr, 2018-03-08:
**PablosScripts commented**

> Please see the example code here:
> 
> https://www.consoletuner.com/forum/viewtopic.php?f=25&t=8005#p58788

AeliusSaionji, 2019-10-07:
I would really like to see this implemented. The documentation doesn't clearly state that this can't be done so I just wasted some time troubleshooting. 

Personally, I wouldn't necessarily need this if the Xbox Elite controller did not automap output buttons 18-21; something I need to unmap in most contexts (especially for switch, where 18 is screenshot).

J2Kbr, 2020-04-21:
Feature implemented on Gtuner IV and Titan Two firmware versions 1.08 (to be released soon).

Added new GPC function `get_port()` to retrieve the port which originated the normalized input value.

Example:
```
main {
    // If BUTTON_16 is active ...
    if(get_actual(BUTTON_16)) {
        // and if the value is comming from USB-B ...
        if(get_port(BUTTON_16) == PORT_USB_B) {
            // don't forward the input to the console.
            set_val(BUTTON_16, 0.0);
        }
    } 
}
```

closed 2020-04-21 by J2Kbr