J2Kbr/TitanTwo #105: Add ability to pair with bluetooth module without using Gtuner IV

PablosScripts, 2018-02-01:
Most other devices allow you to pair using the hardware itself. For example on the XIM you just press the Sync button. 

The problem with needing to use Gtuner IV is it inhibits experimentation and flexibility. I like to test new ideas in regards to different controller setups, and I also like to regularly switch from using a controller on one device to using on another. If I need to use Gtuner IV in order to re-pair, it becomes really troublesome and discourages me from using the cross controller support of Titan Two to it's full potential. Every time I think about using a different setup I think _"ummmm....have to pull out my laptop....too much trouble"_. 

Now I know the T2 doesn't have a dedicated pairing button but this could be achieved in a number of ways, e.g. perhaps by long-pressing one of the profile buttons this could put it into BT searching mode. Or maybe triple tapping. Or perhaps provide us with a script command that we can trigger via a keyboard shortcut.

J2Kbr, 2018-02-20:
Press the Titan Two UP button for 2.5 seconds to enter in Bluetooth pairing mode for 20 seconds. Implemented on firmware revision 0.98B-15. Thank you for the suggestion.

closed 2018-02-20 by J2Kbr

PablosScripts, 2018-02-20:
Thanks. Just a reminder to add this to your guides / documentation so new users are aware:)