J2Kbr/TitanTwo #106: Issue - Script re-running on new report from controller can massively increase CPU usage.

Fen-Ria, 2018-02-02:
Okay, I just noticed whay I believe to be a major issue that seems to relate to having the script re-run on receiving a new report from the controller while using a PS4 controller.

I have a fairly large script, and when I have my PS4 controller plugged in, and then load and run the script, the CPU seems to average at about 30 percent, constantly fluctuating from around 20 to 40 percent, and spikes to 100 percent every now and then.

If I plug the PS4 controller in while the script is running, the Titan Two temporarily disconnects, and when I select it again, the CPU usage is constantly fluctuating between 100 and somehow 111 percent, occassionally dipping down for a fraction of a second, then going right back up.

When I use a wired 360 controller with that same script, it stays consistently at about 10-11 percent, and might spike once or twice when I first run the script.

If I plug the 360 controller in while the script is running, the Titan Two does not disconnect, and the CPU usage spikes several times when first plugged in, but then levels out at about the same CPU usage it would have if I had plugged the controller in, then loaded the script.

I am working on recording the issue right now.

Fen-Ria, 2018-02-02:
I have made two videos.

One has the output port in use, and the other does not.

Test 1 (Output in use):
[https://www.youtube.com/watch?v=_UETizIo9lM](url)

Test 2 (Output not in use):
[https://www.youtube.com/watch?v=uS1iMNP_ihE](url)

Fen-Ria, 2018-02-02:
I have also posted something else the relates to issues with having the script re-run upon recieving a new report from a controller and using a PS4 controller in the Gtuner Github, as I believed it would best fit there due to being a feature request most most likely to be solved by having a tickbox or script code for it.

Fen-Ria, 2018-02-04:
I have recently discovered that the larger or more complex a script is, the bigger the difference gets, to the point that the CPU is always at or around 100 percent when using a PS4 controller, but not when using a 360 controller.

I can attempt to provide a video of it if requested.

J2Kbr, 2018-02-19:
Do you have `while` or `for` loops in the script?
Could you PM me on the forums the script code so I can try reproduce this issue?
Thanks.

Fen-Ria, 2018-03-06:
I have done as requested, and I apologize for the delay.

Fen-Ria, 2018-03-10:
Regarding what you said in [GtunerIV GitHub Issue #198](https://github.com/J2Kbr/GtunerIV/issues/198) about the script finishing the current execution before running again upon receiving a new report from the controller, you had also said in [this forum post](https://www.consoletuner.com/forum/viewtopic.php?f=26&t=7565&p=57023#p57023) "The Titan Two runs the script at least once every 1ms and, if it received a new report from any connected controller, the script runs immediately to ensure the smallest latency possible."

The two statements seem mutually exclusive to me, and the statement about the script re-running immediately caused me to think that it re-runs the script, even if it is already in the middle of running, instead of letting it finish it's current run, causing it to either cancel/interrupt the already running iteration of the script and run a new iteration with the new information, or to run a new iteration of the script with the new information without cancelling the already running one first.

If you could please explain exactly how it works, I would appreciate it, as it would clear up the confusion I have about it.

Fen-Ria, 2018-03-10:
I have found another problem that might be related to the script re-running upon recieving a new report from a controller, but I shall wait till I have more information about how the Titan Two treats new controller reports before I post it. 

The reason is that depending on how it works, I would either need to post the information here due to being related in a similar manner to the issue that you closed for being related, or in a new issue, as it might have an unknown cause.

If you would rather I just post it here, regardless of the potential cause of the issue, please let me know.

J2Kbr, 2018-03-11:
> I tried to figure out how what you had just said about how the script re-runs upon receiving a new report would be different enough from how a script would normally re-run upon finishing a run, but without a new report, that you felt the need to specifically state that it was something to take into consideration in addition to the script running every 1ms.

Sorry if I was not clear in my previous statements. Okay, this is how it works:

1. There are two events that triggers the script execution: (1) a 1ms timeout, (2) a new report from a connected controller was received. If any of this two evaluates to TRUE, the script is executed.

2. If the execution of a script has started, it will run until the end even if a new report form a connected controller is received.

3. Only after the script finishes its execution, the events listed on 1 are checked again.

Fen-Ria, 2018-03-11:
Ah, okay.
Thank you for explaining.

Is it possible that the lack of a 1ms timeout when constantly sending new information to the Titan Two by using a PS4 controller or rapidly pressing buttons could cause some scripts to break if they use a function to keep a combo running, or use a function to apply the effects of a combo to one or more buttons?

I ask this because I've encountered issues with combos incorrectly operating when run in the previously stated manner, and I have determined that when using a wired 360 controller instead of a PS4 controller and leave it idle, the issue almost completely vanishes.

If, however, I rapidly move the thumb-stick or press the buttons on the 360 controller, it causes a similar issue to when I just have a PS4 controller plugged in, albeit to a much smaller extent.

I actually have a small script that I used for testing this, and if requested, I can provide the script and/or a video of the issue.

J2Kbr, 2018-03-11:
yes, please. Send me the small script you have, it will help me better understand the issue. Thanks.

Fen-Ria, 2018-03-12:
I have just sent you the small script via a PM on the forums.

J2Kbr, 2018-03-12:
You are correct, the execution of the combos in your small example will get out of sync if the script is early executed due the receiving of report from any connected controller. Change the function `fAlternate()` to the code below prevent that to happen:

```
void fAlternate() { // Continuously runs both combos and applies the appropriate offset to the Alt_Turbo combo.
                    // It even adjusts the offset if the "ON" and/or "OFF" duration changes while the script is running.
    if(elapsed_time()) {
        if(!Turbo) {
            ALT = (ON + OFF) / 2;
            combo_run(Turbo);
        }
        if(ALT) {
            ALT = ALT - (int16) elapsed_time();
            if(ALT <= 0) {
                combo_run(Alt_Turbo);
            }
        }
    }
}
```

Fen-Ria, 2018-03-13:
So that solves the issue with the combos, but even after updating the Titan Two's firmware to the one released on the 8th, the CPU issue is still there.

Have you managed to make any progress tracking down what might be causing it?

Fen-Ria, 2018-03-13:
Oh, something that would probably be helpful for trying to find out what's causing the CPU usage issue:

If you use a PS4 controller with the first script I sent you, then tap R2, you'll notice that it applies turbo to R2, and you'll see the CPU usage drop like a rock.

If you tap R2 a second time, the turbo is no longer applied, and the CPU usage goes right back to how it was before,

You can also do this with R1, which will apply a different turbo to R1, and again cause the CPU usage to drop. but the CPU usage is higher than if you had tapped R2.

If you tap, L2, you'll notice it applies the Press combo to L2, and the CPU usage ends up lower than if you had tapped R2,

As with R2, both R1, and L2 can be tapped a second time to remove the turbo or press features, and the CPU usage will go right back up.

You can have all multiple buttons and their respective functions active at the same time, and if you have both R2 and R1 use their turbo functions, you'll see the CPU usage goes up by what seems to be about 7% compared to just having R1 active, or about 10% compared to having R2 active.

Having L2 active in addition to R1 and/or R2 does not seem to affect CPU usage in any meaningful way.

J2Kbr, 2018-03-13:
Sorry, I am not finding your full script (I believe to have overwritten it with the smaller code). Please send me again. I will be sure to save in a safe place this time. thanks.

Fen-Ria, 2018-03-13:
I have sent you the script again via a PM on the forums.

Please let me know if you have any questions about the script or need something clarified.

J2Kbr, 2018-03-14:
After checking your script carefully (very complex, hard to make a mental image of its workflow), the high CPU usage seems to be associated with the fAlternate() -> fTurbo() and fAlt_Turbo(). Disabling fAlt_turbo seems to reduce the CPU usage. 

I couldn't find anything technically wrong in your code causing high CPU usage, it just makes lot processing, conditionals, branches, etc.

Instead of manually manage the on/off state of turbo buttons, why not use combos instead? this will surely make the script more optimized and reduce the CPU usage.

Fen-Ria, 2018-03-14:
I have adjusted the script to use combos, and the CPU usage is slightly lower, but there is still a significant difference in usage between the PS4 controller, and the 360 controller.

When idle, the PS4 controller seems to cause an average CPU usage of about 27% based on visual observations, and the the actual CPU usage can change by over 30% in the span of 10ms, and will occasionally spike to 100%, although it seems much more common for it to only change by about 10%.

When idle, the 360 controller seems to cause an average CPU usage of about 10% based on visual observations, and the the actual CPU usage seems to only ever change by about 1% in the span of 10ms,

The combo irregularity still appears in this script, but now it happens approximately every 1.25 seconds, regardless of which controller is used.

Fen-Ria, 2018-03-14:
I can provide the script if requested, and have attempted to explain what each function does via comments in the adjusted script.

I am unfortunately not very good at explaining how exactly the code works, or my logic behind how I set it up which is reflected in the comments in the script, but I am able to provide instructions on how to use the script with a controller if requested.

J2Kbr, 2018-03-14:
Please send me the updated code.

The difference between the PS4 and XB360 controllers is: the PS4 will send a data report every 4ms while the XB360 only sends data reports when a input value changes, if the controller is resting is like the same as not having it connected.

Fen-Ria, 2018-03-15:
I have sent you the updated code via a PM in the forums.

Is it possible that somehow the constant data reports, combined with the script re-running upon receiving a data report, is causing the significant increase in CPU usage?

J2Kbr, 2018-03-16:
what is the purpose of 

`fToggle1();`
and 
`fMultifunction_Toggle1();`

I suppose this is a generic script where you can set turbo mode on many buttons.
So, how do you configure the buttons to have turbo, using the initial value of the variables T1, T2, T3, etc?

Fen-Ria, 2018-03-16:
`fToggle1()` and `fMultifunction_Toggle1()` are both used to set which button will change which T# variable in `fToggle2(int8 b, int16 t)` and `fMultifunction_Toggle2(int8 b, int16 t)` respectively.

In the case of `fToggle1()`, it does not matter what value T# is in order to activate `fToggle2(int8 b, int16 t)`, and `fToggle2(int8 b, int16 t)` will always be active, but will not change any of the T# variables unless a button assigned to a T# is pressed for less than 100ms and released, in which case, it will only change the T# variable assigned to that button.

If you tap R2 for example, it will only change the variable T2 as that is the variable assigned to it by `fToggle1()` for use in `fToggle2(int8 b, int16 t)`.

As as all T# variables are linked to specific buttons, I have made sure that in all cases where a T# variable can be linked to a button, it is always linked to the same button throughout the script, and as such, you could consider T2 to mean "Functionality modifier for R2".

As such, I can use the numeric values of the T# variables to determine if turbo should be active on a specific button or not, and have done so with R2 by using 2 to mean "Auto-Turbo (Inactive)" and 222 to mean "Auto-Turbo (Active)".

If you look at `fToggle1()` and `fToggle2(int8 b, int16 t)`, you can see that `fToggle1()` assigns the R2 button to `b`, and the T2 variable to `t`, so when you tap R2, `fToggle2(int8 b, int16 t)` checks to see what value T2 has, and as it currently has a 2 assigned to it, it then sets T2 to 222, which `fTurbo(int8 v)` sees as a valid condition to apply the `Turbo` combo to R2 and the blue LED.

If you then tap the button a second time, using the same process as above, `fToggle2(int8 b, int16 t)` sees that T2 has a value of 222, and sets it to 2, which is not a valid condition for `fTurbo(int8 v)` to apply the `Turbo` combo to R2 and the blue LED, effectively turning the combo off for that button.

A similar method is applied to the buttons in `fMultifunction_Toggle1()`, but requires that their corresponding T# value first be either 3 or 5 in order for `fMultifunction_Toggle1()` to use the button and it's corresponding T# with fMultifunction_Toggle2(int8 b, int16 t), which can assign one of two values depending on if a button was tapped then held, or just held.

Here is a list of the T# variables and their corresponding buttons, as well as what the currently used numeric values mean:
> T1 = R1
> T2 = R2
> T3 = R3
> T4 = L1
> T5 = L2
> T6 = L3
> T7 = Triangle
> T8 = Circle
> T9 = Cross
> T10 = Square
> 
> 0 = No features
> 1 = Turbo (Only when the button is being pressed)
> 2 = Auto-Turbo (Inactive, Tap to turn on.)
> 222 = Auto-Turbo (Active, Tap to turn off.)
> 3 = Multifunction Toggle Set 1 (Inactive, Tap then hold button to turn auto-turbo on, or just hold the button to turn auto-hold on.)
> 33 = Multifunction Toggle Set 1 (Auto-Hold Active, Tap to turn off.)
> 333 = Multifunction Toggle Set 1 (Auto-Turbo Active, Tap to turn off.)
> 4 = Auto-Hold (Inactive, Tap to turn on.)
> 44 = Auto-Hold (Active, Tap to turn off.)
> 5 = Multifunction Toggle Set 2 (Inactive, Tap then hold button to turn auto-alt turbo on, or just hold the button to turn auto-hold on.)
> 55 = Multifunction Toggle Set 2 (Auto-Hold Active, Tap to turn off.)
> 555 = Multifunction Toggle Set 2 (Auto-Alt Turbo Active, Tap to turn off.)
> 6 = Auto-Alt Turbo (Inactive, Tap to turn on.)
> 666 = Auto-Alt Turbo (Active, Tap to turn off.)
> 7 = Alt Turbo (Only when the button is being pressed)

Please let me know if you have any more questons about my script, or if you need something I've said to be further clarified.

Fen-Ria, 2018-03-19:
Something else that might possibly be related to the issues I have been having:

When using the script with a PS4 controller, when something happens that would activate the rumble on the controller for a short time, the rumble occasionally gets stuck on until something that would normally activate the rumble happens again.

I have determined that it has to do with using the script and the PS4 controller due to the issue not happening when the Titan Two is set to slot 0 when using the PS4 controller, nor when using the script with a 360 controller.

Also, do you know what could be causing the combo irregularity that appears approximately every 1.25 seconds?

It seems way to regular for it to be linked to a representation error in Gtuner IV's graph.

Fen-Ria, 2018-03-22:
No idea what happened, but it seems like the problem has vanished.

closed 2018-03-22 by Fen-Ria