J2Kbr/TitanTwo #108: Input Device Lockup

BrandGSX, 2018-02-10:
Console: PS4
Device: Titan Two, FW 0.98.1.11
Peripherals: Razer Deathadder 2013, Dell SK-3205 keyboard, Ri-Japan 8GB MicroSD HC1 Class 10

Running: Input Translation for mouse and keyboard only, no GPC scripts. 

Configuration is Standalone. No connection to PC
TT output to console
TT Input B to DS4 controller
TT Input A to Mouse
TT Prog port to Keyboard through OTG cable

Detail:
When using the TT one of the inputs devices will stop responding to input commands. Usually the controller.
If using the mouse and keyboard after a while the controller will not accept input commands. I usually use the controller for menu navigation and will not notice the error until I need to use the DS4.
In some instances I use the Controller and the mouse only and when the controller stops working the current input values are frozen to the values at the time of the lockup. Say STICK_2_Y is at -85 input it will stay that way until the DS4 is unplugged and plugged back in.. This was confirmed with GTunerIV in the device monitor when the TT is connected to the PC. It also does this with no connection to the PC.
During the lockups the input on the working devices can sometimes stick or freeze to a high value. This is usually not noticeable unless navigating a menu. Using ASWD for menu navigation it will sometimes stick and scroll all the way through the menu or until another key is pressed.

Note 1 when the controller stops accepting input it still keep authenticating the Titan Two for the PS4.
Note 2 If I use KM capture for the mouse and keyboard i have no problems at all with the controller locking up.

J2Kbr, 2018-02-19:
Thank you for the detailed report. It really helped me to head to a possible cause for this issue. 

To confirm, could you please do one last test?

Use the controller on `Input-B`, the mouse on `PROG-OTG` port and nothing on `Input-A`. Please let me know if the controller still freezes, or not.

Thank you.

J2Kbr, 2018-02-21:
Please also let me know if you have a headset attached to the PS4 controller.

BrandGSX, 2018-02-21:
With the controller on Input-B, the mouse on PROG-OTG port and nothing on Input-A and it did not lock up. After this seemed OK I tried plugging the keyboard into Input-A and it still seemed to work. Is keeping the mouse in the Program port the solution?

I do not use a headset through the PS4 controller.

J2Kbr, 2018-02-27:
Hi, this issue was very possible fixed with the 0.98B-18 update. Please comment if not. Thanks.

closed 2018-02-27 by J2Kbr

BrandGSX, 2018-03-02:
OK so after trying the new update today (been busy) it seems to be working ok. No lockup and no input sticks so it seems lie this did the trick. With less than an hour of game time with the update I will set some more and report if it comes back in any way.