J2Kbr/TitanTwo #117: Memory Slot change per XB controller works only once

monty35, 2018-03-03:
Hi, as the title says the Memory Slot change per XB controller (LB/ RB with XB Button) works only once, afterwards no effect if the buttons are pressed again. The hard buttons at T2 (Up/Down) works fine.

Firmware 0.98.18

J2Kbr, 2018-03-08:
I tested here with a PS3, PS4 and XB1 controllers and worked as expected. The script that loads with the first `[LB|RB]+[XB]` buttons press does any change on controller mapping? and, if removing the SD card it works? Thanks.

monty35, 2018-03-10:
Hi, it seems to related to the remapping function, therefore no technical issue here. But maybe you can advise in the instruction that the LB/RB & XB feature only works if no remapping is activated for these buttons in one of the saved slots.

closed 2018-03-10 by monty35