J2Kbr/TitanTwo #123: USB Headsets are not supported

M4V3R1C8, 2018-03-13:
Using a USB headset such as the Turtle Beach 400x and 420x causes the Titan Two to disconnect (scripts continue to work after re-connecting controller) and even reset (scripts no longer work after re-connecting controller and requires the user to toggle between scripts to restart them.)

J2Kbr, 2018-03-13:
Please inform the gaming system (PS3/XB1), the controller and if the controller is connected wired or wirelessly to the Titan Two.

M4V3R1C8, 2018-03-14:
Both Xbox One and PS4 with a wired controller, regardless of cross over.

J2Kbr, 2018-03-14:
Thank you for informing.

The crossover of controller headsets is not supported. The audio data packets and polling time are very different between the consoles, making very hard perform this conversion.

When using crossover feature you can have your preferred controller on Input-A of the Titan Two, and the original controller for the console on Input-B. With the original controller you can use the headset and still play using your preferred controller.

closed 2018-03-14 by J2Kbr

M4V3R1C8, 2018-03-15:
That is correct, you can play the game with either controller while using your preferred controller in either input as long as you use the wired headset on the console specific controller. However, if you use a console specific USB headset on the respective console (without crossover) the Titan Two will reset several times in a 10 minute time interval.

cr33dk1ller, 2018-03-15:
i believe this is the very same issue i have been having as well ever since i got the t2. i never imagined it could be because i was using a usb headset. will do some testing this weekend.