J2Kbr/TitanTwo #131: Recent firmware update causes some scripts to crash the Titan Two

Fen-Ria, 2018-03-21:
I have a script I had been using for a while, and I last updated the script on the 16th, at which point it was working fine.

After I updated the firmware this morning, the script is now unusable, as the Titan Two almost immediately crashes upon loading it.

Here is the script that now crashes the Titan Two:
`#pragma METAINFO("Warframe Script", 0, 95, "Fenria")

#define NOT_USE			0
#define DZ_CIRCLE		101

#define PIO_NONE		0
#define PIO_AUTO		0
#define PIO_PS4			4
#define PIO_PS3			1
#define PIO_XB1			5
#define PIO_XB360		2
#define PIO_WII			3
#define PIO_HID			6
#define PIO_SWITCH		7

#define PS4_PS			BUTTON_1
#define PS4_TOUCH		BUTTON_18	//BUTTON_2
#define PS4_OPTIONS		BUTTON_3
#define PS4_R1			BUTTON_4
#define PS4_R2			BUTTON_5
#define PS4_R3			BUTTON_6
#define PS4_L1			BUTTON_7
#define PS4_L2			BUTTON_8
#define PS4_L3			BUTTON_9
#define PS4_UP			BUTTON_10
#define PS4_DOWN		BUTTON_11
#define PS4_LEFT		BUTTON_12
#define PS4_RIGHT		BUTTON_13
#define PS4_TRIANGLE	BUTTON_14
#define PS4_CIRCLE		BUTTON_15
#define PS4_CROSS		BUTTON_16
#define PS4_SQUARE		BUTTON_17
#define PS4_SHARE		BUTTON_2	//BUTTON_18
#define PS4_TOUCH1		BUTTON_19
#define PS4_TOUCH2		BUTTON_20

#define PS4_RX			STICK_1_X
#define PS4_RY			STICK_1_Y
#define PS4_LX			STICK_2_X
#define PS4_LY			STICK_2_Y

#define PS4_TOUCH1X		POINT_1_X
#define PS4_TOUCH1Y		POINT_1_Y
#define PS4_TOUCH2X		POINT_2_X
#define PS4_TOUCH2Y		POINT_2_Y

#define PS4_ACCX		ACCEL_1_X
#define PS4_ACCY		ACCEL_1_Y
#define PS4_ACCZ		ACCEL_1_Z

#define PS4_GYROX		GYRO_1_X
#define PS4_GYROY		GYRO_1_Y
#define PS4_GYROZ		GYRO_1_Z

#define TRACE_1			ACCEL_2_X
#define TRACE_2			ACCEL_2_Y
#define TRACE_3			ACCEL_2_Z
#define TRACE_4			GYRO_1_X
#define TRACE_5			GYRO_1_Y
#define TRACE_6			GYRO_1_Z

int16 LC = 0;
int16 LC1 = 0;
int16 LC2 = 0;
int16 LC3 = 0;
[... 415 lines cut ...]
	if(mid != NOT_USE)	{
		int16 val_s = (val >= 0) ? 1 : -1;
		val *= val_s;
		if(val <= mid) val = (val * 50) / mid;
		else val = ((50 * (val - mid)) / (100 - mid)) + 50;
		val *= val_s;
	}
	if(sen != NOT_USE)	{
		val = (val * sen) / 100;
	}
	set_val(id, clamp(val, -100, 100));
	return;
}

void output_protocol(int16 id) {
	port_disconnect();
	switch(id)	{
		case PIO_PS3: port_connect(PORT_USB_C, PROTOCOL_PS3); break;
		case PIO_XB360: port_connect(PORT_USB_C, PROTOCOL_XB360); break;
		case PIO_PS4: port_connect(PORT_USB_C, PROTOCOL_PS4); break;
		case PIO_XB1: port_connect(PORT_USB_C, PROTOCOL_XB1); break;
		case PIO_SWITCH: port_connect(PORT_USB_C, PROTOCOL_SWITCH); break;
	}
	return;
}`

Fen-Ria, 2018-03-21:
Here is the script that now crashes the Titan Two:
```
#pragma METAINFO("Warframe Script", 0, 95, "Fenria")

#define NOT_USE			0
#define DZ_CIRCLE		101

#define PIO_NONE		0
#define PIO_AUTO		0
#define PIO_PS4			4
#define PIO_PS3			1
#define PIO_XB1			5
#define PIO_XB360		2
#define PIO_WII			3
#define PIO_HID			6
#define PIO_SWITCH		7

#define PS4_PS			BUTTON_1
#define PS4_TOUCH		BUTTON_18	//BUTTON_2
#define PS4_OPTIONS		BUTTON_3
#define PS4_R1			BUTTON_4
#define PS4_R2			BUTTON_5
#define PS4_R3			BUTTON_6
#define PS4_L1			BUTTON_7
#define PS4_L2			BUTTON_8
#define PS4_L3			BUTTON_9
#define PS4_UP			BUTTON_10
#define PS4_DOWN		BUTTON_11
#define PS4_LEFT		BUTTON_12
#define PS4_RIGHT		BUTTON_13
#define PS4_TRIANGLE	BUTTON_14
#define PS4_CIRCLE		BUTTON_15
#define PS4_CROSS		BUTTON_16
#define PS4_SQUARE		BUTTON_17
#define PS4_SHARE		BUTTON_2	//BUTTON_18
#define PS4_TOUCH1		BUTTON_19
#define PS4_TOUCH2		BUTTON_20

#define PS4_RX			STICK_1_X
#define PS4_RY			STICK_1_Y
#define PS4_LX			STICK_2_X
#define PS4_LY			STICK_2_Y

#define PS4_TOUCH1X		POINT_1_X
#define PS4_TOUCH1Y		POINT_1_Y
#define PS4_TOUCH2X		POINT_2_X
#define PS4_TOUCH2Y		POINT_2_Y

#define PS4_ACCX		ACCEL_1_X
#define PS4_ACCY		ACCEL_1_Y
#define PS4_ACCZ		ACCEL_1_Z

#define PS4_GYROX		GYRO_1_X
#define PS4_GYROY		GYRO_1_Y
#define PS4_GYROZ		GYRO_1_Z

#define TRACE_1			ACCEL_2_X
#define TRACE_2			ACCEL_2_Y
#define TRACE_3			ACCEL_2_Z
#define TRACE_4			GYRO_1_X
#define TRACE_5			GYRO_1_Y
#define TRACE_6			GYRO_1_Z

int16 LC = 0;
int16 LC1 = 0;
int16 LC2 = 0;
int16 LC3 = 0;
int16 DZ = 10;
int16 RS = 100;
int16 LS = 100;
[... 413 lines cut ...]
		int16 val_s = (val >= 0) ? 1 : -1;
		val *= val_s;
		if(val <= mid) val = (val * 50) / mid;
		else val = ((50 * (val - mid)) / (100 - mid)) + 50;
		val *= val_s;
	}
	if(sen != NOT_USE)	{
		val = (val * sen) / 100;
	}
	set_val(id, clamp(val, -100, 100));
	return;
}

void output_protocol(int16 id) {
	port_disconnect();
	switch(id)	{
		case PIO_PS3: port_connect(PORT_USB_C, PROTOCOL_PS3); break;
		case PIO_XB360: port_connect(PORT_USB_C, PROTOCOL_XB360); break;
		case PIO_PS4: port_connect(PORT_USB_C, PROTOCOL_PS4); break;
		case PIO_XB1: port_connect(PORT_USB_C, PROTOCOL_XB1); break;
		case PIO_SWITCH: port_connect(PORT_USB_C, PROTOCOL_SWITCH); break;
	}
	return;
}
````

J2Kbr, 2018-03-21:
Changes reverted in revision 40

closed 2018-03-21 by J2Kbr

Fen-Ria, 2018-03-21:
Thank you for fixing it.