J2Kbr/TitanTwo #134: Gtuner IV (Titan Two) Wish List

Tech-G33K, 2018-03-28:
Options I would like to see back in Gtuner IV such as:

-Enable remote control of slot switch on device / G8 Alternate.
-More memory Slots on the SD Card.

This topic will help the development of Gtuner IV Titan Two.

**Please share your Wish List of what you would like to see in the future releases.**

norge420, 2018-04-11:
hmm titan one i thought you could do this by guide+ start next slot  and guide+ select for previous

Tech-G33K, 2018-04-12:
This issue is for **Gtuner IV (Titan Two)** only discussion, Thanks.

NeoTechni, 2018-06-10:
> -More memory Slots on the SD Card.

Agreed, since it can display letters, A-Z would be nice too.

J2Kbr, 2018-11-14:
Additional memory slots will be provided via this suggestion: https://github.com/J2Kbr/GtunerIV/issues/237

closed 2018-11-14 by J2Kbr