J2Kbr/TitanTwo #137: [Still Happening] Frequent disconnects with Xbox One Wireless adapter

misterpyrrhuloxia, 2018-04-08:
I am on the last firmware for the T2 as I just updated on April 8th, 2018. And still, while playing BF1 on Xbox with the wireless adapter and Elite Controller, when there are massive explosions in game, my controller disconnects from the adapter and then I have to spend several minutes just trying to get it to reconnect, only to have it happen again when the very next massive explosion happens. This is getting to be unplayable like this.

J2Kbr, 2018-04-14:
The firmware file for the Xbox Wireless Adapter was updated in the Titan Two version 0.98B-45 (to be released during this next week). It is expected improvements on the connection stability.

_PS.: Gtuner IV also requires to be updated and the SD Card reconfigured._

misterpyrrhuloxia, 2018-04-19:
I upgraded to T2 firmware Ver 0.98B-45 and re-enabled force feedback and once again while playing tonight for the first time since upgrading, my Elite controller disconnected from the wireless adapter. It doesn't seem to have been fixed.

J2Kbr, 2018-04-19:
Have you also re configured the SD Card with the new Xbox Wireless adapter firmware?

misterpyrrhuloxia, 2018-04-19:
After updating the firmware, I clicked on the button in Gtuner to Configure the sd card. It took at least 30 seconds and pushed a bunch of files to the T2. Was I supposed to reformat the sd card before reconfiguring it?

J2Kbr, 2018-04-19:
Thank you for confirming, Issue reopened.

misterpyrrhuloxia, 2018-04-20:
Does this same problem happen while using the new Xbox wireless adapter? Because I only have the first gen adapter.

misterpyrrhuloxia, 2018-05-15:
Any progress on this issue yet?

misterpyrrhuloxia, 2018-06-08:
Any word on this yet? My controller disconnects daily. It's really annoying.

J2Kbr, 2018-06-09:
Improvements on this issue implemented on FW 0.98B-50, to be released soon.

closed 2018-06-09 by J2Kbr

misterpyrrhuloxia, 2018-06-10:
How soon?

syntactic-salt, 2018-12-06:
I know this is closed, but will this ever be fixed? I bought the titan two for this specific use case and it's practically unusable with the constant disconnects.

J2Kbr, 2018-12-06:
Please be sure to have the latest firmware installed in your device.

syntactic-salt, 2018-12-09:
> Please be sure to have the latest firmware installed in your device.

I do. Latest firmware, latest gtuner, and reconfigured the sd card.

syntactic-salt, 2018-12-10:
How do I get support for this device?

misterpyrrhuloxia, 2018-12-10:
Here's what Jefferson had me try months ago if the adapter refuses to pair properly with the controller. This works for me every time even if both of the normal pairing methods A and B [from here](https://www.consoletuner.com/wiki/index.php?id=t2:usage_guides:controllers:xboxone#wireless_adapter) don't work for me. If I were you, I'd save the below steps somewhere where you can easily access it on the future. I just store them in a [Google Keep](https://keep.google.com) note.

```
Pair instructions if it won't pair properly:

1.   Unplug the T2 from the Xbox and unplug the adapter from the T2.
2.   Plug the T2 into the Xbox using the PROG port.
3.   Orange PROG LED should light up on the T2.
4.   Plug controller into input port A on the T2.
5.   Input LED should turn green.
6.   Power off the controller by unplugging the controller from the T2 and pulling out the battery and then putting the battery back in.
7.   Plug the adapter into input port A on T2.
8.   Input LED should turn green.
9.   Turn on controller - It should pair successfully with dim controller LED.
10.  Power off the controller by pulling out the battery and then putting the battery back in.
11.  Unplug T2 from Xbox.
12.  Plug T2 into the Xbox using the OUTPUT port.
13.  Input LED should turn green.
14.  Turn on controller.
15.  Controller should have successfully paired with adapter.
```

syntactic-salt, 2018-12-10:
I appreciate your reply, but that's a pretty ridiculous set of steps for something that should just work. I'd really like to hear from the @J2Kbr about this issue.