J2Kbr/TitanTwo #138: Lost Connection from Xbox One Elite to Wireless Dongle (FAT Model) in Port A

DebuRanger, 2018-04-17:
Hi,

So Xbox One Elite -> Microsoft Wireless Dongle (FAT) -> Titan 2 -> PS4.

I tested the Xbox One Elite to the wireless adapter to Gtuner and it was working fine.
Buttons were flashing in the screen etc.

After I'd say about 1 hour or maybe 2, it dropped connect to the dongle.
My character started spinning on screen and I couldn't resync to the wireless adapter.
PS4 controller was on Port B for the Authentication and was powered on.
PS4 mode was using the USB for controllers.

I ended going wired the rest of the time.

Game was Destiny 2 on PS4 (Not that it matters) and I was using the paddles on the Elite.

misterpyrrhuloxia, 2018-04-19:
This is happening to me too, only on Xbox while playing BF1. I reported it here:  #137 

He closed my issue already saying that a fix was happening on T2 firmware Ver 0.98B-45, which was incoming at the time, but the issue is still happening to me even after updating to that firmware version.

DebuRanger, 2018-04-19:
Seems like the Issue isn't fixed yet I guess.  I pulled the update maybe last week or so.  So I guess it is still there.  I'll have to check the version later.

J2Kbr, 2018-06-09:
Seems the disconnection is related with FFB data packets. Improvements related to this on FW 0.98B-50, to be released in the upcoming days.

closed 2018-06-09 by J2Kbr