J2Kbr/TitanTwo #144: Support for Hyperkin Duke XB1 Controller

adrianluff, 2018-05-28:
Please add support for the Hyperkin Duke XB1 controller. USB descriptor from GTunerIV:
[Hyperkin_Duke_USB_Descriptors.txt](https://github.com/J2Kbr/TitanTwo/files/2043437/Hyperkin_Duke_USB_Descriptors.txt)

NeoTechni, 2018-06-10:
It already supports it. The only problem is, What would be the share button triggers Touch Click.

J2Kbr, 2018-06-10:
The descriptors shows to be a standard Xbox One controller. Thank you NeoTechni to confirming this model is already supported.

The PS4 Share button can be programmatically activated via script, or remapped to any existing button via Input Translator (or script).

```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
00 02        bcdUSB: 2.00
FF           bDeviceClass: 255 (FF)
FF           bDeviceSubClass: 255 (FF)
FF           bDeviceProtocol: 255 (FF)
40           bMaxPacketSize0: 64
24 2E        idVendor: 2E24
52 06        idProduct: 0652
01 01        bcdDevice: 1.01
01           iManufacturer: 1
02           iProduct: 2
03           iSerialNumber: 3
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
60 00        wTotalLength: 96
03           bNumInterfaces: 3
01           bConfigurationValue: 1
00           iConfiguration: 0
A0           bmAttributes: 160 (Bus Powered, Remote Wakeup)
FA           MaxPower: 500mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
FF             bInterfaceClass: 255 (FF)
47             bInterfaceSubClass: 71 (47)
D0             bInterfaceProtocol: 208 (D0)
00             iInterface: 0

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
01               bEndpointAddress: 1 OUT
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
04               bInterval: 4

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
81               bEndpointAddress: 1 IN
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
04               bInterval: 4

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
00             bAlternateSetting: 0
00             bNumEndpoints: 0
FF             bInterfaceClass: 255 (FF)
47             bInterfaceSubClass: 71 (47)
D0             bInterfaceProtocol: 208 (D0)
00             iInterface: 0

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
01             bAlternateSetting: 1
02             bNumEndpoints: 2
FF             bInterfaceClass: 255 (FF)
47             bInterfaceSubClass: 71 (47)
D0             bInterfaceProtocol: 208 (D0)
00             iInterface: 0

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
08               bEndpointAddress: 8 OUT
01               bmAttributes: 1 (Isochronous, None, Data)
E4 00            wMaxPacketSize: 228
01               bInterval: 1

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
87               bEndpointAddress: 7 IN
01               bmAttributes: 1 (Isochronous, None, Data)
E4 00            wMaxPacketSize: 228
01               bInterval: 1

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
02             bInterfaceNumber: 2
00             bAlternateSetting: 0
00             bNumEndpoints: 0
FF             bInterfaceClass: 255 (FF)
47             bInterfaceSubClass: 71 (47)
D0             bInterfaceProtocol: 208 (D0)
00             iInterface: 0

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
02             bInterfaceNumber: 2
01             bAlternateSetting: 1
02             bNumEndpoints: 2
FF             bInterfaceClass: 255 (FF)
47             bInterfaceSubClass: 71 (47)
D0             bInterfaceProtocol: 208 (D0)
00             iInterface: 0

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
03               bEndpointAddress: 3 OUT
02               bmAttributes: 2 (Bulk, None, Data)
40 00            wMaxPacketSize: 64
00               bInterval: 0

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
83               bEndpointAddress: 3 IN
02               bmAttributes: 2 (Bulk, None, Data)
40 00            wMaxPacketSize: 64
00               bInterval: 0

[STRING 1 DESCRIPTOR]
DDR

[STRING 2 DESCRIPTOR]
Hyperkin Duke

[STRING 3 DESCRIPTOR]
0000C2FE715DDBC8

```

closed 2018-06-10 by J2Kbr