J2Kbr/TitanTwo #145: Support for the Gamesir G5 controller

NeoTechni, 2018-06-04:
This [thing](https://sites.google.com/site/neotechni/Home/pc#TOC-GameSir-G5) has a lot of buttons. The Titan Two won't detect the 4 extra triggers or the 8 extra face buttons. The left/right sticks have weird ranges too (left: -100 to 14436, right: -1700 to 15636, both neutral: -25700) and of course, since it's a third party mobile controller the button mapping is all wrong.

[USB DESCRIPTOR]
12 01 10 01 00 00 00 08 AC 05 7A 05 02 03 06 07 00 01 09 02 42 00 02 01 00 A0 FA 09 04 00 00 02 03 00 00 00 09 21 11 01 00 01 22 A1 00 07 05 02 03 40 00 10 07 05 82 03 10 00 10 09 04 01 00 01 03 01 02 00 09 21 10 01 00 01 22 34 00 07 05 81 03 10 00 04 

[STRING 6 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 06 04 09 00 0E 
0E 03 58 00 69 00 61 00 6F 00 6A 00 69 00 

[STRING 7 DESCRIPTOR]
00 00 00 02 00 00 02 01 80 06 03 07 04 09 00 16 
16 03 47 00 61 00 6D 00 65 00 73 00 69 00 72 00 2D 00 47 00 35 00 

[INTERFACE 0 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 00 00 A1 
05 01 09 05 A1 01 85 C5 05 01 09 05 A1 00 09 30 09 31 15 00 25 FF 35 00 45 FF 95 02 75 08 81 02 C0 A1 00 09 32 09 35 15 00 25 FF 35 00 45 FF 95 02 75 08 81 02 C0 05 02 09 05 A1 00 09 C5 09 C4 15 00 25 FF 35 00 45 FF 95 02 75 08 81 02 C0 05 09 19 01 29 10 15 00 25 01 35 00 45 01 75 01 95 10 81 02 05 01 19 01 29 10 15 00 25 01 35 00 45 01 75 01 95 10 81 02 05 01 09 39 15 01 25 08 35 00 46 3B 10 66 0E 00 75 04 95 01 81 42 75 04 95 01 81 03 A1 02 85 05 09 04 75 08 95 01 91 02 C0 C0 

[INTERFACE 1 HID DESCRIPTOR]
00 00 00 02 00 00 02 01 81 06 22 00 00 01 00 34 
05 01 09 02 A1 01 09 01 A1 00 85 01 09 30 09 31 09 38 75 08 95 03 15 81 25 7F 81 06 15 00 25 01 05 09 19 01 29 03 75 01 95 03 81 02 75 05 95 01 81 03 C0 C0

J2Kbr, 2018-06-20:
Adding the descriptors decoded for future reference.

```
[USB DESCRIPTOR]
12           bLength: 18
01           bDescriptorType: 1 - DEVICE
10 01        bcdUSB: 1.10
00           bDeviceClass: 0 (00)
00           bDeviceSubClass: 0 (00)
00           bDeviceProtocol: 0 (00)
08           bMaxPacketSize0: 8
AC 05        idVendor: 05AC
7A 05        idProduct: 057A
02 03        bcdDevice: 3.02
06           iManufacturer: 6
07           iProduct: 7
00           iSerialNumber: 0
01           bNumConfigurations: 1

09           bLength: 9
02           bDescriptorType: 2 - CONFIGURATION
42 00        wTotalLength: 66
02           bNumInterfaces: 2
01           bConfigurationValue: 1
00           iConfiguration: 0
A0           bmAttributes: 160 (Bus Powered, Remote Wakeup)
FA           MaxPower: 500mA

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
00             bInterfaceNumber: 0
00             bAlternateSetting: 0
02             bNumEndpoints: 2
03             bInterfaceClass: 3 (03)
00             bInterfaceSubClass: 0 (00)
00             bInterfaceProtocol: 0 (00)
00             iInterface: 0

09               bLength: 9
21               bDescriptorType: 33 - HID
11 01            bcdHID: 1.11
00               bCountryCode: 0
01               bNumDescriptors: 1
22               bDescriptorType: 34
A1 00            wDescriptorLength: 161

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
02               bEndpointAddress: 2 OUT
03               bmAttributes: 3 (Interrupt, None, Data)
40 00            wMaxPacketSize: 64
10               bInterval: 16

07               bLength: 7
05               bDescriptorType: 5 - ENDPOINT
82               bEndpointAddress: 2 IN
03               bmAttributes: 3 (Interrupt, None, Data)
10 00            wMaxPacketSize: 16
10               bInterval: 16

09             bLength: 9
04             bDescriptorType: 4 - INTERFACE
01             bInterfaceNumber: 1
00             bAlternateSetting: 0
01             bNumEndpoints: 1
03             bInterfaceClass: 3 (03)
01             bInterfaceSubClass: 1 (01)
02             bInterfaceProtocol: 2 (02)
00             iInterface: 0

[... 109 lines cut ...]
09 01          (LOCAL)  USAGE              0x00010001 Pointer (CP=Physical Collection) 
A1 00          (MAIN)   COLLECTION         0x00 Physical (Usage=0x00010001: Page=Generic Desktop Page, Usage=Pointer, Type=CP)
85 01            (GLOBAL) REPORT_ID          0x01 (1) 
09 30            (LOCAL)  USAGE              0x00010030 X (DV=Dynamic Value) 
09 31            (LOCAL)  USAGE              0x00010031 Y (DV=Dynamic Value) 
09 38            (LOCAL)  USAGE              0x00010038 Wheel (DV=Dynamic Value) 
75 08            (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field 
95 03            (GLOBAL) REPORT_COUNT       0x03 (3) Number of fields 
15 81            (GLOBAL) LOGICAL_MINIMUM    0x81 (-127) 
25 7F            (GLOBAL) LOGICAL_MAXIMUM    0x7F (127) 
81 06            (MAIN)   INPUT              0x00000006 (3 fields x 8 bits) 0=Data 1=Variable 1=Relative 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
15 00            (GLOBAL) LOGICAL_MINIMUM    0x00 (0) 
25 01            (GLOBAL) LOGICAL_MAXIMUM    0x01 (1) 
05 09            (GLOBAL) USAGE_PAGE         0x0009 Button Page 
19 01            (LOCAL)  USAGE_MINIMUM      0x00090001 Button 1 Primary/trigger (MULTI=Selector, On/Off, Momentary, or One Shot) 
29 03            (LOCAL)  USAGE_MAXIMUM      0x00090003 Button 3 Tertiary (MULTI=Selector, On/Off, Momentary, or One Shot) 
75 01            (GLOBAL) REPORT_SIZE        0x01 (1) Number of bits per field 
95 03            (GLOBAL) REPORT_COUNT       0x03 (3) Number of fields
81 02            (MAIN)   INPUT              0x00000002 (3 fields x 1 bit) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
75 05            (GLOBAL) REPORT_SIZE        0x05 (5) Number of bits per field 
95 01            (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields 
81 03            (MAIN)   INPUT              0x00000003 (1 field x 5 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
C0             (MAIN)   END_COLLECTION     Physical
C0           (MAIN)   END_COLLECTION     Application
```

J2Kbr, 2018-06-20:
Note, the range values of the sticks should be correct for the next firmware update (0.98B-53).

closed 2018-12-03 by J2Kbr

sam0383, 2022-01-11:
Hello, 
how to start the joystick?
after pressing the reset button, it stopped turning on (Does not respond to pressing the power button, and the indexer does not light up). You need a full firmware of the joystick using OTG, since it does not connect to android devices via bluetooth. when connected to the phone with a cable, he does not see it. When connected to a Windows computer, it appears as a device but does not work.

requires primary firmware for the joystick