J2Kbr/TitanTwo #154: Elite Controller disconnected from adapter on Xbox - cannot pair again

misterpyrrhuloxia, 2018-07-28:
So for months now my controller will disconnect from the wireless adapter for seemingly no reason. It used to disconnect every day when there were massive explosions in BF1. That was caused by the issue that has since been supposedly fixed. But even though since that fix I've experienced less disconnections, my controller still disconnects from the wireless adapter for no reason, even without any explosions in-game or anything that would cause my controller to disconnect.

I'm losing my mind here. This has now happened two days in a row. The controller disconnecting wouldn't be such a big deal if I could at least get the controller to freakin pair back with the adapter right away. But no. Last night, it took me 30 minutes of me fiddling to get it to pair again. Tonight, after it disconnected, I've been fighting with it for over an hour, still not able to get them to pair. 

Details:
I can pair the controller directly to the Xbox. - - Works ok. 
I can connect the Elite Controller directly to the T2 with a cable, and it authorizes and works ok. 
But when I power cycle the T2, then plug in the wireless adapter into Input-A, and try to pair the controller to the adapter, at first they look like they're pairing; but actually what happens is the T2 never authorizes with the Xbox, and after the controller light is solid for a few seconds, it starts slowly flashing again as if it never paired in the first place. And I know that the elite controller isn't authorizing with the Xbox because when it looks like they're paired for a few seconds there while the controller led is solid, the T2 just continues to constantly flash "UA". It sometimes authorizes, but the controller isn't actually paired with the adapter. The controller led will be solid for a few seconds and then start flashing slowly. Sometimes it doesn't even flash slowly right away, but the T2 and therefore the Xbox, never respond to input from the controller in this state. Another thing I've noticed is that whenever it has this issue with pairing and it's solid for a few seconds (making me think it was successful), the led on the controller is brighter than it normally is when it's paired properly and working.

I'm on the latest version of T2 firmware from July 19th, 2018.

Here's a video I recorded of the steps I did and the issue occurring:
https://youtu.be/5keqmIzL-5w

misterpyrrhuloxia, 2018-08-01:
[This issue has been solved.](https://www.consoletuner.com/forum/viewtopic.php?f=27&t=9613&p=68097&wpureload=1#p68224)

J2Kbr, 2018-10-15:
Firmware version 0.98B-65 brings more improvements on connections stability of Xbox Wireless Adapter (old[big]/new[small]).

closed 2018-10-15 by J2Kbr